Ysurac/openmptcprouter #2326: Warning issues about create image for NXP imx6

atyouspace, 2022-04-16:
## Expected Behavior

I follow the guide in "https://github.com/Ysurac/openmptcprouter/wiki/Create-image-for-unsupported-platform". 
`
$ git clone https://github.com/Ysurac/openmptcprouter.git
$ cd openmptcprouter
$ git checkout master
$ OMR_TARGET="NXPimx6" OMR_FEED_SRC="master" ./build.sh
`
Lots of WARNING are here.
`
Target NXPimx6 not found !
hint: Using 'master' as the name for the initial branch. This default branch name
hint: is subject to change. To configure the initial branch name to use in all
hint: of your new repositories, which will suppress this warning, call:
hint: 
hint: 	git config --global init.defaultBranch <name>
hint: 
hint: Names commonly chosen instead of 'master' are 'main', 'trunk' and
hint: 'development'. The just-created branch can be renamed via this command:
hint: 
hint: 	git branch -m <name>
Initialized empty Git repository in /home/osboxes/openmptcprouter/NXPimx6/source/.git/
...
...
WARNING: Makefile 'package/utils/busybox/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a build dependency on 'libpam', which does not exist
WARNING: Makefile 'package/kernel/linux/Makefile' has a dependency on 'kmod-cdrom', which does not exist
WARNING: Makefile 'package/kernel/linux/Makefile' has a dependency on 'kmod-cdrom', which does not exist
WARNING: Makefile 'package/kernel/linux/Makefile' has a dependency on 'kmod-cdrom', which does not exist
WARNING: Makefile 'package/kernel/linux/Makefile' has a dependency on 'kmod-cdrom', which does not exist
WARNING: Makefile 'package/kernel/linux/Makefile' has a dependency on 'kmod-ledtrig-audio', which does not exist
WARNING: Makefile 'package/feeds/openmptcprouter/openmptcprouter-full/Makefile' has a dependency on 'luci-proto-modemmanager', which does not exist
WARNING: Makefile 'package/feeds/openmptcprouter/openmptcprouter-full/Makefile' has a dependency on 'kmod-tcp-bbr2', which does not exist
WARNING: Makefile 'package/feeds/openmptcprouter/openmptcprouter-mini/Makefile' has a dependency on 'luci-proto-modemmanager', which does not exist
WARNING: Makefile 'package/utils/policycoreutils/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/policycoreutils/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/policycoreutils/Makefile' has a build dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/wmt/Makefile' has a dependency on 'kmod-mt6625l-wlan-gen2', which does not exist
Installing package 'macremapper' from packages
Done
Target NXPimx6 not found ! You have to configure and compile your kernel manually.

`

## Specifications

  - OpenMPTCProuter version: *master
  - OpenMPTCProuter platform: imx6

Ysurac, 2022-04-16:
No real problems here, only "luci-proto-manager" missing is strange.

atyouspace, 2022-04-16:
> No real problems here, only "luci-proto-manager" missing is strange.

Wouldn't these warnings cause potential problems later?
If not, for next, what should I do? 

Compile the kernel manually?
`cd NXPimx6/source`
`make menuconfig`

Choose "imx6" as the hardware.

`make -j1 V=s`

is that right?

atyouspace, 2022-04-17:
Hi @Ysurac ,

I try to create image for RPi4 platform which is included in the openmptcprouter.
However, the same WARNING issues are there.
I paste all the log for your review and check the problem.

What should I do next? Thank you.

===
~/openmptcprouter$ OMR_TARGET="rpi4" OMR_FEED_SRC="master" ./build.sh
hint: Using 'master' as the name for the initial branch. This default branch name
hint: is subject to change. To configure the initial branch name to use in all
hint: of your new repositories, which will suppress this warning, call:
hint: 
hint: 	git config --global init.defaultBranch <name>
hint: 
hint: Names commonly chosen instead of 'master' are 'main', 'trunk' and
hint: 'development'. The just-created branch can be renamed via this command:
hint: 
hint: 	git branch -m <name>
Initialized empty Git repository in /home/osboxes/openmptcprouter/rpi4/source/.git/
remote: Enumerating objects: 598303, done.
remote: Counting objects: 100% (39/39), done.
remote: Compressing objects: 100% (39/39), done.
remote: Total 598303 (delta 0), reused 38 (delta 0), pack-reused 598264
Receiving objects: 100% (598303/598303), 217.79 MiB | 966.00 KiB/s, done.
Resolving deltas: 100% (403510/403510), done.
From https://github.com/openwrt/openwrt
 * [new branch]            lede-17.01    -> origin/lede-17.01
 * [new branch]            master        -> origin/master
 * [new branch]            openwrt-18.06 -> origin/openwrt-18.06
 * [new branch]            openwrt-19.07 -> origin/openwrt-19.07
 * [new branch]            openwrt-21.02 -> origin/openwrt-21.02
 * [new branch]            openwrt-22.03 -> origin/openwrt-22.03
 * [new tag]               reboot        -> reboot
 * [new tag]               v17.01.2      -> v17.01.2
 * [new tag]               v17.01.3      -> v17.01.3
 * [new tag]               v17.01.4      -> v17.01.4
 * [new tag]               v17.01.5      -> v17.01.5
 * [new tag]               v17.01.6      -> v17.01.6
 * [new tag]               v17.01.7      -> v17.01.7
 * [new tag]               v18.06.0      -> v18.06.0
 * [new tag]               v18.06.0-rc1  -> v18.06.0-rc1
 * [new tag]               v18.06.0-rc2  -> v18.06.0-rc2
 * [new tag]               v18.06.1      -> v18.06.1
 * [new tag]               v18.06.2      -> v18.06.2
 * [new tag]               v18.06.3      -> v18.06.3
 * [new tag]               v18.06.4      -> v18.06.4
 * [new tag]               v18.06.5      -> v18.06.5
 * [new tag]               v18.06.6      -> v18.06.6
 * [new tag]               v18.06.7      -> v18.06.7
 * [new tag]               v18.06.8      -> v18.06.8
 * [new tag]               v18.06.9      -> v18.06.9
 * [new tag]               v19.07.0      -> v19.07.0
 * [new tag]               v19.07.0-rc1  -> v19.07.0-rc1
 * [new tag]               v19.07.0-rc2  -> v19.07.0-rc2
 * [new tag]               v19.07.1      -> v19.07.1
 * [new tag]               v19.07.2      -> v19.07.2
 * [new tag]               v19.07.3      -> v19.07.3
 * [new tag]               v19.07.4      -> v19.07.4
 * [new tag]               v19.07.5      -> v19.07.5
 * [new tag]               v19.07.6      -> v19.07.6
 * [new tag]               v19.07.7      -> v19.07.7
 * [new tag]               v19.07.8      -> v19.07.8
 * [new tag]               v19.07.9      -> v19.07.9
 * [new tag]               v21.02.0      -> v21.02.0
 * [new tag]               v21.02.0-rc1  -> v21.02.0-rc1
 * [new tag]               v21.02.0-rc2  -> v21.02.0-rc2
 * [new tag]               v21.02.0-rc3  -> v21.02.0-rc3
 * [new tag]               v21.02.0-rc4  -> v21.02.0-rc4
[... 406 lines cut ...]
WARNING: Makefile 'package/utils/policycoreutils/Makefile' has a build dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/wmt/Makefile' has a dependency on 'kmod-mt6625l-wlan-gen2', which does not exist
#
# configuration written to .config
#
WARNING: Makefile 'package/utils/busybox/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a build dependency on 'libpam', which does not exist
WARNING: Makefile 'package/kernel/linux/Makefile' has a dependency on 'kmod-cdrom', which does not exist
WARNING: Makefile 'package/kernel/linux/Makefile' has a dependency on 'kmod-cdrom', which does not exist
WARNING: Makefile 'package/kernel/linux/Makefile' has a dependency on 'kmod-cdrom', which does not exist
WARNING: Makefile 'package/kernel/linux/Makefile' has a dependency on 'kmod-cdrom', which does not exist
WARNING: Makefile 'package/kernel/linux/Makefile' has a dependency on 'kmod-ledtrig-audio', which does not exist
WARNING: Makefile 'package/feeds/openmptcprouter/openmptcprouter-full/Makefile' has a dependency on 'luci-proto-modemmanager', which does not exist
WARNING: Makefile 'package/feeds/openmptcprouter/openmptcprouter-mini/Makefile' has a dependency on 'luci-proto-modemmanager', which does not exist
WARNING: Makefile 'package/utils/policycoreutils/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/policycoreutils/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/policycoreutils/Makefile' has a build dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/wmt/Makefile' has a dependency on 'kmod-mt6625l-wlan-gen2', which does not exist
time: target/linux/prereq#0.11#0.01#0.15
 make[1] world
 make[2] tools/compile
...
...
===

Ysurac, 2022-04-17:
You can ignore all warnings, and no problem later.
Yes you have to a "make menuconfig" in source/<arch> and a "make -j$(nproc)" (-j1 V=s is for debug, to see errors)

atyouspace, 2022-04-17:
> You can ignore all warnings, and no problem later.

Some hours later, the compiling is finished, but there is an error. Have you met this before?

ERROR: package/feeds/openmptcprouter/v2ray-core failed to build.

I will try to `make -j1 V=s`.

atyouspace, 2022-04-17:
Hi @Ysurac 

I am trying to `make -j1 V=s`.
Stop here:

.config:1495: warning: override: DEFAULT_OLIA changes choice state
*
* Restart config...
*
*
* Core Netfilter Configuration
*
Netfilter ingress support (NETFILTER_INGRESS) [Y/n/?] y
Netfilter NFNETLINK interface (NETFILTER_NETLINK) [M/y/?] m
Netfilter NFACCT over NFNETLINK interface (NETFILTER_NETLINK_ACCT) [N/m/y/?] n
Netfilter NFQUEUE over NFNETLINK interface (NETFILTER_NETLINK_QUEUE) [M/y/?] m
Netfilter LOG over NFNETLINK interface (NETFILTER_NETLINK_LOG) [M/y/?] m
Netfilter OSF over NFNETLINK interface (NETFILTER_NETLINK_OSF) [N/m/y/?] n
Netfilter connection tracking support (NF_CONNTRACK) [M/n/y/?] m
Netdev packet logging (NF_LOG_NETDEV) [N/m/y/?] n
Connection mark tracking support (NF_CONNTRACK_MARK) [Y/?] y
Connection tracking zones (NF_CONNTRACK_ZONES) [Y/n/?] y
Supply CT list in procfs (OBSOLETE) (NF_CONNTRACK_PROCFS) [Y/n/?] y
Connection tracking events (NF_CONNTRACK_EVENTS) [Y/n/?] y
  **Register multiple callbacks to ct events (NF_CONNTRACK_CHAIN_EVENTS) [N/y/?] (NEW)**

What's should I do next?

Ysurac, 2022-04-17:
You can press enter for default (N here).
But to compile only a package, you can do a "make -j1 V=s package/feeds/openmptcprouter/v2ray-core/compile"

atyouspace, 2022-04-17:
Hi @Ysurac 

Stop here.

I guess it is because the package（golang related） was not downloaded successfully? How to download manually?

../../infra/conf/services.go:6:2: github.com/golang/protobuf@v1.5.2: Get "https://proxy.golang.org/github.com/golang/protobuf/@v/v1.5.2.zip": dial tcp 172.217.160.113:443: i/o timeout
../../common/serial/typed_message.go:7:2: github.com/golang/protobuf@v1.5.2: Get "https://proxy.golang.org/github.com/golang/protobuf/@v/v1.5.2.zip": dial tcp 172.217.160.113:443: i/o timeout
../../transport/internet/websocket/connection.go:12:2: github.com/gorilla/websocket@v1.4.2: Get "https://proxy.golang.org/github.com/gorilla/websocket/@v/v1.4.2.zip": dial tcp 142.251.43.17:443: i/o timeout
../../infra/conf/api.go:6:2: github.com/jhump/protoreflect@v1.9.0: Get "https://proxy.golang.org/github.com/jhump/protoreflect/@v/v1.9.0.zip": dial tcp 172.217.160.113:443: i/o timeout
../../infra/conf/api.go:7:2: github.com/jhump/protoreflect@v1.9.0: Get "https://proxy.golang.org/github.com/jhump/protoreflect/@v/v1.9.0.zip": dial tcp 172.217.160.113:443: i/o timeout
../../app/dns/nameserver_quic.go:13:2: github.com/lucas-clemente/quic-go@v0.23.0: Get "https://proxy.golang.org/github.com/lucas-clemente/quic-go/@v/v0.23.0.zip": dial tcp 172.217.160.113:443: i/o timeout
../../transport/internet/system_listener.go:8:2: github.com/pires/go-proxyproto@v0.6.1: Get "https://proxy.golang.org/github.com/pires/go-proxyproto/@v/v0.6.1.zip": dial tcp 142.251.43.17:443: i/o timeout
../../common/antireplay/replayfilter.go:7:2: github.com/seiflotfy/cuckoofilter@v0.0.0-20201222105146-bc6005554a0c: Get "https://proxy.golang.org/github.com/seiflotfy/cuckoofilter/@v/v0.0.0-20201222105146-bc6005554a0c.zip": dial tcp 172.217.160.113:443: i/o timeout
../../app/browserforwarder/forwarder.go:14:2: github.com/v2fly/BrowserBridge@v0.0.0-20210430233438-0570fc1d7d08: Get "https://proxy.golang.org/github.com/v2fly/%21browser%21bridge/@v/v0.0.0-20210430233438-0570fc1d7d08.zip": dial tcp 142.251.43.17:443: i/o timeout
../../common/platform/securedload/embeddedhash.go:10:2: github.com/v2fly/VSign@v0.0.0-20201108000810-e2adc24bf848: Get "https://proxy.golang.org/github.com/v2fly/%21v%21sign/@v/v0.0.0-20201108000810-e2adc24bf848.zip": dial tcp 142.251.43.17:443: i/o timeout
../../common/platform/securedload/embeddedhash.go:11:2: github.com/v2fly/VSign@v0.0.0-20201108000810-e2adc24bf848: Get "https://proxy.golang.org/github.com/v2fly/%21v%21sign/@v/v0.0.0-20201108000810-e2adc24bf848.zip": dial tcp 142.251.43.17:443: i/o timeout
../../common/antireplay/bloomring.go:6:2: github.com/v2fly/ss-bloomring@v0.0.0-20210312155135-28617310f63e: Get "https://proxy.golang.org/github.com/v2fly/ss-bloomring/@v/v0.0.0-20210312155135-28617310f63e.zip": dial tcp 142.251.43.17:443: i/o timeout
../../app/router/condition.go:9:2: go.starlark.net@v0.0.0-20210901212718-87f333178d59: Get "https://proxy.golang.org/go.starlark.net/@v/v0.0.0-20210901212718-87f333178d59.zip": dial tcp 172.217.160.113:443: i/o timeout
../../app/router/condition.go:10:2: go.starlark.net@v0.0.0-20210901212718-87f333178d59: Get "https://proxy.golang.org/go.starlark.net/@v/v0.0.0-20210901212718-87f333178d59.zip": dial tcp 172.217.160.113:443: i/o timeout
../../proxy/shadowsocks/config.go:11:2: golang.org/x/crypto@v0.0.0-20210817164053-32db794688a5: Get "https://proxy.golang.org/golang.org/x/crypto/@v/v0.0.0-20210817164053-32db794688a5.zip": dial tcp 172.217.160.113:443: i/o timeout
../../proxy/shadowsocks/config.go:12:2: golang.org/x/crypto@v0.0.0-20210817164053-32db794688a5: Get "https://proxy.golang.org/golang.org/x/crypto/@v/v0.0.0-20210817164053-32db794688a5.zip": dial tcp 172.217.160.113:443: i/o timeout
../../proxy/vmess/encoding/auth.go:8:2: golang.org/x/crypto@v0.0.0-20210817164053-32db794688a5: Get "https://proxy.golang.org/golang.org/x/crypto/@v/v0.0.0-20210817164053-32db794688a5.zip": dial tcp 172.217.160.113:443: i/o timeout
../../common/protocol/dns/io.go:7:2: golang.org/x/net@v0.0.0-20210903162142-ad29c8ab022f: Get "https://proxy.golang.org/golang.org/x/net/@v/v0.0.0-20210903162142-ad29c8ab022f.zip": dial tcp 172.217.160.113:443: i/o timeout
../../app/dns/nameserver_quic.go:15:2: golang.org/x/net@v0.0.0-20210903162142-ad29c8ab022f: Get "https://proxy.golang.org/golang.org/x/net/@v/v0.0.0-20210903162142-ad29c8ab022f.zip": dial tcp 172.217.160.113:443: i/o timeout
../../transport/internet/http/hub.go:14:2: golang.org/x/net@v0.0.0-20210903162142-ad29c8ab022f: Get "https://proxy.golang.org/golang.org/x/net/@v/v0.0.0-20210903162142-ad29c8ab022f.zip": dial tcp 172.217.160.113:443: i/o timeout
../../transport/internet/filelocker_other.go:9:2: golang.org/x/sys@v0.0.0-20210903071746-97244b99971b: Get "https://proxy.golang.org/golang.org/x/sys/@v/v0.0.0-20210903071746-97244b99971b.zip": dial tcp 172.217.160.113:443: i/o timeout
../../app/commander/commander.go:13:2: google.golang.org/grpc@v1.41.0: Get "https://proxy.golang.org/google.golang.org/grpc/@v/v1.41.0.zip": dial tcp 172.217.160.113:443: i/o timeout
../../transport/internet/grpc/dial.go:13:2: google.golang.org/grpc@v1.41.0: Get "https://proxy.golang.org/google.golang.org/grpc/@v/v1.41.0.zip": dial tcp 172.217.160.113:443: i/o timeout
../../app/log/command/config_grpc.pb.go:8:2: google.golang.org/grpc@v1.41.0: Get "https://proxy.golang.org/google.golang.org/grpc/@v/v1.41.0.zip": dial tcp 172.217.160.113:443: i/o timeout
[... cut ...]
    ERROR: package/feeds/openmptcprouter/v2ray-core failed to build.
make[2]: *** [package/Makefile:116: package/feeds/openmptcprouter/v2ray-core/compile] Error 1
make[2]: Leaving directory '/home/osboxes/openmptcprouter/NXPimx6/source'
make[1]: *** [package/Makefile:110: /home/osboxes/openmptcprouter/NXPimx6/source/staging_dir/target-arm_cortex-a9+neon_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/osboxes/openmptcprouter/NXPimx6/source'
make: *** [/home/osboxes/openmptcprouter/NXPimx6/source/include/toplevel.mk:230: world] Error 2

Ysurac, 2022-04-18:
All downloaded files are in source/dl directory.

atyouspace, 2022-04-18:
> All downloaded files are in source/dl directory.
Hi @Ysurac 

Thanks for your reply.
I am not sure why these errors happened.
Should I download them manually again?

atyouspace, 2022-04-18:
Is it possible that the kernel version of Debian system on virtualbox is 5.10 instead of kernel 5.4?

Ysurac, 2022-04-18:
The problem doesn't seems to be related to host system, you can try to download a failed URL to check.

atyouspace, 2022-04-18:
> You can press enter for default (N here).
> But to compile only a package, you can do a "make -j1 V=s package/feeds/openmptcprouter/v2ray-core/compile"

I see the "Makefile" in "package/feeds/openmptcprouter/v2ray-core/".
I found this download URL is not working any more. Because I paste it into the web browser address, "404" is there.
"PKG_SOURCE_URL:=https://github.com/v2fly/v2ray-core/archive/refs/tags/"

So is it a bug? @Ysurac

Ysurac, 2022-04-18:
Not it's not a bug, because URL is $PKG_SOURCE_URl$PKG_SOURCE
Remember that a snapshot is build at each commit, so when there is such bug no image are build. As I try to keep all as open as possible, build are visible here: https://github.com/Ysurac/openmptcprouter-feeds/actions (when there is a problem, all fail, this happen when I do a big mistake)

atyouspace, 2022-04-24:
Thanks for all. This issue has been solved.

closed 2022-04-24 by atyouspace