Ysurac/openmptcprouter #2327: Bootloop with Sophos XG85W

tuanbombt, 2022-04-16:
## Expected Behavior
I am facing a bootloop error when installing openmptcprouter with sophos xg85w.

Configuration information:
Intel(R) Atom(TM) Processor E3930 @ 1.30GHz
RAM: 2GB
eMMC: 8GB.

After installing I had to reconfigure grub.cfg to boot because they don't have display ports (dvi, vga).

Then I see errors related to ACPI..., I can't get the logs because they reboot too fast.

## Current Behavior
Don't any log.

## Specifications

  - OpenMPTCProuter version: openmptcprouter-v0.58.5-r0+16336-b36068d35d-x86
  - OpenMPTCProuter VPS version: 0.1026
  - OpenMPTCProuter VPS provider: Lightsail
  - OpenMPTCProuter platform: x86

tuanbombt, 2022-04-17:
@Ysurac I can't boot, could you please tell me, what is the problem that is happening here?

Ysurac, 2022-04-17:
This looks like a problem with i915. Can be hardware or driver problem.

tuanbombt, 2022-04-17:
I can boot with other openwrt x86 have firmware i915 in folder (lib/firmware/i915).

If i want build other firmware have i915, what should i do now?

tuanbombt, 2022-04-17:
You can see here:
[image]

Ysurac, 2022-04-18:
You can try latest beta https://download.openmptcprouter.com/release/v0.59beta3-5.4/ or latest snapshot https://download.openmptcprouter.com/develop/5.4/x86_64/

tuanbombt, 2022-04-18:
It's work, thank you very much

closed 2022-04-18 by tuanbombt