Ysurac/openmptcprouter #2328: Weird performance behaviour on netcup

finnschi, 2022-04-17:
So i am sucessfully running a VPS with netcup, according to speedtest-cli its around 600/600 ish from the vps and I get around 600/250 from home which isnt bad at all. 

But now I stepped it up to a root server with netcup and it suddenly dropped to like 150/50  ? I go back to the vps abd all is good again, the root server has a 2.5G wan link and speedtest cli shows 2000/2000 . 

latency is the same. 

testing the speed using a openspeedtest setup on another root server in the same datacenter. 

What could be the difference here? Reason is I have a single docsis 1G line 2 dsl 100 lines  and a LTE connection so there is quiet a bit of performance left on the table and I was hoping for more speed with the root server. 

CPU wise i have a i5 10200U or something in my router and 16G of ram so total absolute overkill, the vps also has like 4 epyc cores are 8G

Ysurac, 2022-04-18:
As it's working with same config on the VPS, check Status->System log if there is no problems.
New server was rebooted ? No problem in /var/log/* on this server ?

closed 2022-07-22 by github-actions[bot]

Adorfer, 2022-07-22:
it's hard to tell what may be the issue since the booked netcup products seem to be secret(?). 
and yes there are caveats to fall into. 
e.g. "RS 1000 G9 a1" from netcup provides 1GBit/s+ via docsis + 5G from the OMR.