Ysurac/openmptcprouter #2329: How do you open port ranges in Google Cloud Platform?

dooley1981, 2022-04-17:
[Google Cloud Platform](https://cloud.google.com/) port range 0-65535 in UDP, TCP and ICMP must be opened

I guess it must be open to work with Openmptcprouter?

Ysurac, 2022-04-17:
Port 65000-65535 are used by OpenMPTCProuter internal communications, others ports can be used if you redirect ports.
VPS part have a firewall installed and configured by default that block all port from 0 to 65000 (Only ICMP is authorized)

dooley1981, 2022-04-17:
Okay so I shouldn't have to do anything for it to work with this software correct?

Ysurac, 2022-04-17:
You only have to open ports, as they are closed by default on GCP

closed 2022-04-18 by dooley1981

Ysurac, 2025-05-08:
Try to connect to https;//<vpsip>:65500/ and check via ```ps aux | grep omr``` if omr-admin is running.