Ysurac/openmptcprouter #2330: Connection is not stable and has high pings

mstrolo, 2022-04-17:
## Expected Behavior

Bonding is connected to the server and 3 wans.

## Current Behavior

the bond is not stable high pings and request time outs.

## Specifications

  - OpenMPTCProuter version: 
  - [Powered by openmptcprouter v0.58.5 r0+16336-b36068d35d](https://github.com/ysurac/openmptcprouter)
  - OpenMPTCProuter VPS version: 
  - 
[image]

  - OpenMPTCProuter VPS provider:

  - OpenMPTCProuter platform: 
x86_64

[image]

mstrolo, 2022-04-17:
i see this in the logs:

[image]

[image]

mstrolo, 2022-04-17:
this what happens when i run speed test:

[image]

Ysurac, 2022-04-17:
Check in log the error, why the VPN is down, should be something like ping timeout on some IP.
Set also the more stable connection as master.

mstrolo, 2022-04-17:
the VDSL is the master

closed 2022-07-22 by github-actions[bot]