Ysurac/openmptcprouter #2332: Debian 10

dooley1981, 2022-04-18:
My server has Debian 9.3, Can I install the Debian 10 script?

Ysurac, 2022-04-18:
The script update Debian 9 to Debian 10 before installing needed packages

dooley1981, 2022-04-18:
by using this script?     wget -O - https://www.openmptcprouter.com/server/debian9-x86_64.sh | sh

Ysurac, 2022-04-18:
Yes, it's in fact same script for all supported Linux distribution.

dooley1981, 2022-04-18:
I am getting the certificate error so I ran the apt-get update && apt-get upgrade.  It didn't finish and it gave me this message.

cron (3.0pl1-128+deb9u2) stretch-security; urgency=medium

  * As a reasonable protective measure, crontabs are now limited to 10000 lines
    in length per crontab.
    The maintainers find it very unlikely that longer crontabs exist; however,
    if you do have a use case, please file a bug report with a brief rationale,
    and we will consider raising this limit.

dooley1981, 2022-04-20:
Check user...
Check Linux version...
Check architecture...
Check virtualized environment
Check about broken packages...
Check if OpenMPTCProuter VPS is already installed...
Update mode
Remove lock and update packages list...
E: Command line option --allow-releaseinfo-change is not understood in combination with the other options

Now it thinks its already installed. anyway to remove it to start over?

dooley1981, 2022-04-20:
fix it, got a better VPS

closed 2022-04-20 by dooley1981