Ysurac/openmptcprouter #2333: lib32gcc1 is not available, when create image for unsupported platform

atyouspace, 2022-04-19:
## Expected Behavior
You need to be under Linux with all tools needed to compile OpenWrt. This should be enough for Debian, as root:

apt-get install busybox curl rsync build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev libz-dev patch unzip zlib1g-dev lib32gcc1 libc6-dev-i386 subversion flex uglifyjs git-core gcc-multilib p7zip p7zip-full msmtp libssl-dev texinfo libglib2.0-dev xmlto qemu-utils upx libelf-dev autoconf automake libtool autopoint device-tree-compiler wget tar

## Current Behavior
Package lib32gcc1 is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or is only available from another source
However the following packages replace it:
  lib32gcc-s1

E: Package 'lib32gcc1' has no installation candidate
root:~# apt-cache search lib32gcc1
lib32gcc-s1-amd64-cross - GCC support library (32 bit Version)
lib32gcc-s1-s390x-cross - GCC support library (32 bit Version)
lib32gcc-s1-mips64-cross - GCC support library (32 bit Version)
lib32gcc-s1-mips64el-cross - GCC support library (32 bit Version)
lib32gcc-s1-mips64r6-cross - GCC support library (32 bit Version)
lib32gcc-s1-mips64r6el-cross - GCC support library (32 bit Version)
lib32gcc-s1-ppc64-cross - GCC support library (32 bit Version)
lib32gcc-s1-sparc64-cross - GCC support library (32 bit Version)
lib32gcc-s1-x32-cross - GCC support library (32 bit Version)

## Specifications

- host: Debian 11 Bullseye, Kernel 5.10.0-13-amd64
- OpenMPTCProuter platform: imx6

Ysurac, 2022-04-19:
The doc was for Debian 10, I added the version on the page.
So as indicated by apt, for Debian 11 use ```lib32gcc-s1```

atyouspace, 2022-04-19:
I switch to Debian 10. It's done.
Thanks.

closed 2022-04-19 by atyouspace