Ysurac/openmptcprouter #2334: OMR-Bypass with Mac address

voarsh2, 2022-04-19:
## Expected Behavior
Game blocks me because of "VPN".
Set a computer in OMR-Bypass via MAC in the main eth0 to bypass. I expect not to be using the VPN tunnel on the computer with that Mac address.

## Current Behavior

My requests to the game still get blocked

## Specifications

  - OpenMPTCProuter version: v0.58.5 r0+16336-b36068d35d 
  - OpenMPTCProuter VPS version: Version 0.1026 5.4.100-mptcp 
  - OpenMPTCProuter VPS provider: DigitalOcean
  - OpenMPTCProuter platform:  X86_64

Kalimeiro, 2022-04-21:
omr-bypass is not working for domain bypass since beta 4.

Kalimeiro, 2022-04-21:
makes some test and i can confirm :

- domains bypass not working.
- ip and networking bypass not working.
- asn bypass not working.
- protocoles and services bypass not working.

Ysurac, 2022-04-21:
Fixed in latest commit for v0.59, snapshot is building

voarsh2, 2022-06-01:
> Fixed in latest commit for v0.59, snapshot is building

Am I missing something here? https://www.openmptcprouter.com/download shows no new version to flash. I am on v0.58.5 r0+16336-b36068d35d. The only thing I see is in /patch/ folder with [luci-occitan.patch](https://github.com/Ysurac/openmptcprouter/blob/develop/patches/luci-occitan.patch) recently changed. Where can I get the latest patch for my version?

closed 2022-09-05 by github-actions[bot]