Ysurac/openmptcprouter #2335: Building Golang and V2ray has errors

atyouspace, 2022-04-20:
## Current Behavior

I try Debian 10 to create image for imx6 again.
Building Golang and V2ray has errors. I have tried some methods, but the problem is still here.

I appreciate your help.

Finding targets
go: downloading google.golang.org/protobuf v1.27.1
go: downloading github.com/golang/protobuf v1.5.2
go: downloading golang.org/x/sys v0.0.0-20210903071746-97244b99971b
go: downloading google.golang.org/grpc v1.41.0
go: downloading github.com/lucas-clemente/quic-go v0.23.0
go: downloading golang.org/x/net v0.0.0-20210903162142-ad29c8ab022f
go: downloading go.starlark.net v0.0.0-20210901212718-87f333178d59
go: downloading inet.af/netaddr v0.0.0-20210903134321-85fa6c94624e
go: downloading golang.org/x/crypto v0.0.0-20210817164053-32db794688a5
go: downloading github.com/gorilla/websocket v1.4.2
go: downloading github.com/pires/go-proxyproto v0.6.1
go: downloading github.com/jhump/protoreflect v1.9.0
go: downloading github.com/seiflotfy/cuckoofilter v0.0.0-20201222105146-bc6005554a0c
go: downloading github.com/v2fly/ss-bloomring v0.0.0-20210312155135-28617310f63e
go: downloading github.com/v2fly/BrowserBridge v0.0.0-20210430233438-0570fc1d7d08
go: downloading github.com/v2fly/VSign v0.0.0-20201108000810-e2adc24bf848

Building targets
go: downloading google.golang.org/protobuf v1.27.1
go: downloading github.com/golang/protobuf v1.5.2
go: downloading golang.org/x/sys v0.0.0-20210903071746-97244b99971b
go: downloading google.golang.org/grpc v1.41.0
go: downloading github.com/lucas-clemente/quic-go v0.23.0
go: downloading golang.org/x/net v0.0.0-20210903162142-ad29c8ab022f
go: downloading go.starlark.net v0.0.0-20210901212718-87f333178d59
go: downloading inet.af/netaddr v0.0.0-20210903134321-85fa6c94624e
go: downloading golang.org/x/crypto v0.0.0-20210817164053-32db794688a5
go: downloading github.com/gorilla/websocket v1.4.2
go: downloading github.com/pires/go-proxyproto v0.6.1
go: downloading github.com/jhump/protoreflect v1.9.0
go: downloading github.com/seiflotfy/cuckoofilter v0.0.0-20201222105146-bc6005554a0c
go: downloading github.com/v2fly/ss-bloomring v0.0.0-20210312155135-28617310f63e
go: downloading github.com/v2fly/BrowserBridge v0.0.0-20210430233438-0570fc1d7d08
go: downloading github.com/v2fly/VSign v0.0.0-20201108000810-e2adc24bf848
../../infra/conf/services.go:6:2: github.com/golang/protobuf@v1.5.2: Get "https://proxy.golang.org/github.com/golang/protobuf/@v/v1.5.2.zip": dial tcp 172.217.163.49:443: i/o timeout
../../common/serial/typed_message.go:7:2: github.com/golang/protobuf@v1.5.2: Get "https://proxy.golang.org/github.com/golang/protobuf/@v/v1.5.2.zip": dial tcp 172.217.163.49:443: i/o timeout
../../transport/internet/websocket/connection.go:12:2: github.com/gorilla/websocket@v1.4.2: Get "https://proxy.golang.org/github.com/gorilla/websocket/@v/v1.4.2.zip": dial tcp 172.217.160.113:443: i/o timeout
../../infra/conf/api.go:6:2: github.com/jhump/protoreflect@v1.9.0: Get "https://proxy.golang.org/github.com/jhump/protoreflect/@v/v1.9.0.zip": dial tcp 142.251.43.17:443: i/o timeout
../../infra/conf/api.go:7:2: github.com/jhump/protoreflect@v1.9.0: Get "https://proxy.golang.org/github.com/jhump/protoreflect/@v/v1.9.0.zip": dial tcp 142.251.43.17:443: i/o timeout
../../app/dns/nameserver_quic.go:13:2: github.com/lucas-clemente/quic-go@v0.23.0: Get "https://proxy.golang.org/github.com/lucas-clemente/quic-go/@v/v0.23.0.zip": dial tcp 172.217.163.49:443: i/o timeout
../../transport/internet/system_listener.go:8:2: github.com/pires/go-proxyproto@v0.6.1: Get "https://proxy.golang.org/github.com/pires/go-proxyproto/@v/v0.6.1.zip": dial tcp 142.251.43.17:443: i/o timeout
../../common/antireplay/replayfilter.go:7:2: github.com/seiflotfy/cuckoofilter@v0.0.0-20201222105146-bc6005554a0c: Get "https://proxy.golang.org/github.com/seiflotfy/cuckoofilter/@v/v0.0.0-20201222105146-bc6005554a0c.zip": dial tcp 172.217.160.81:443: i/o timeout
../../app/browserforwarder/forwarder.go:14:2: github.com/v2fly/BrowserBridge@v0.0.0-20210430233438-0570fc1d7d08: Get "https://proxy.golang.org/github.com/v2fly/%21browser%21bridge/@v/v0.0.0-20210430233438-0570fc1d7d08.zip": dial tcp 172.217.160.81:443: i/o timeout
../../common/platform/securedload/embeddedhash.go:10:2: github.com/v2fly/VSign@v0.0.0-20201108000810-e2adc24bf848: Get "https://proxy.golang.org/github.com/v2fly/%21v%21sign/@v/v0.0.0-20201108000810-e2adc24bf848.zip": dial tcp 172.217.160.81:443: i/o timeout
../../common/platform/securedload/embeddedhash.go:11:2: github.com/v2fly/VSign@v0.0.0-20201108000810-e2adc24bf848: Get "https://proxy.golang.org/github.com/v2fly/%21v%21sign/@v/v0.0.0-20201108000810-e2adc24bf848.zip": dial tcp 172.217.160.81:443: i/o timeout
../../common/antireplay/bloomring.go:6:2: github.com/v2fly/ss-bloomring@v0.0.0-20210312155135-28617310f63e: Get "https://proxy.golang.org/github.com/v2fly/ss-bloomring/@v/v0.0.0-20210312155135-28617310f63e.zip": dial tcp 172.217.160.81:443: i/o timeout
../../app/router/condition.go:9:2: go.starlark.net@v0.0.0-20210901212718-87f333178d59: Get "https://proxy.golang.org/go.starlark.net/@v/v0.0.0-20210901212718-87f333178d59.zip": dial tcp 172.217.163.49:443: i/o timeout
../../app/router/condition.go:10:2: go.starlark.net@v0.0.0-20210901212718-87f333178d59: Get "https://proxy.golang.org/go.starlark.net/@v/v0.0.0-20210901212718-87f333178d59.zip": dial tcp 172.217.163.49:443: i/o timeout
../../proxy/shadowsocks/config.go:11:2: golang.org/x/crypto@v0.0.0-20210817164053-32db794688a5: Get "https://proxy.golang.org/golang.org/x/crypto/@v/v0.0.0-20210817164053-32db794688a5.zip": dial tcp 172.217.163.49:443: i/o timeout
../../proxy/shadowsocks/config.go:12:2: golang.org/x/crypto@v0.0.0-20210817164053-32db794688a5: Get "https://proxy.golang.org/golang.org/x/crypto/@v/v0.0.0-20210817164053-32db794688a5.zip": dial tcp 172.217.163.49:443: i/o timeout
../../proxy/vmess/encoding/auth.go:8:2: golang.org/x/crypto@v0.0.0-20210817164053-32db794688a5: Get "https://proxy.golang.org/golang.org/x/crypto/@v/v0.0.0-20210817164053-32db794688a5.zip": dial tcp 172.217.163.49:443: i/o timeout
../../common/protocol/dns/io.go:7:2: golang.org/x/net@v0.0.0-20210903162142-ad29c8ab022f: Get "https://proxy.golang.org/golang.org/x/net/@v/v0.0.0-20210903162142-ad29c8ab022f.zip": dial tcp 172.217.163.49:443: i/o timeout
../../app/dns/nameserver_quic.go:15:2: golang.org/x/net@v0.0.0-20210903162142-ad29c8ab022f: Get "https://proxy.golang.org/golang.org/x/net/@v/v0.0.0-20210903162142-ad29c8ab022f.zip": dial tcp 172.217.163.49:443: i/o timeout
../../transport/internet/http/hub.go:14:2: golang.org/x/net@v0.0.0-20210903162142-ad29c8ab022f: Get "https://proxy.golang.org/golang.org/x/net/@v/v0.0.0-20210903162142-ad29c8ab022f.zip": dial tcp 172.217.163.49:443: i/o timeout
../../transport/internet/filelocker_other.go:9:2: golang.org/x/sys@v0.0.0-20210903071746-97244b99971b: Get "https://proxy.golang.org/golang.org/x/sys/@v/v0.0.0-20210903071746-97244b99971b.zip": dial tcp 172.217.163.49:443: i/o timeout
../../app/commander/commander.go:13:2: google.golang.org/grpc@v1.41.0: Get "https://proxy.golang.org/google.golang.org/grpc/@v/v1.41.0.zip": dial tcp 172.217.163.49:443: i/o timeout
../../transport/internet/grpc/dial.go:13:2: google.golang.org/grpc@v1.41.0: Get "https://proxy.golang.org/google.golang.org/grpc/@v/v1.41.0.zip": dial tcp 172.217.163.49:443: i/o timeout
../../app/log/command/config_grpc.pb.go:8:2: google.golang.org/grpc@v1.41.0: Get "https://proxy.golang.org/google.golang.org/grpc/@v/v1.41.0.zip": dial tcp 172.217.163.49:443: i/o timeout
../../transport/internet/grpc/dial.go:14:2: google.golang.org/grpc@v1.41.0: Get "https://proxy.golang.org/google.golang.org/grpc/@v/v1.41.0.zip": dial tcp 172.217.163.49:443: i/o timeout
../../transport/internet/grpc/dial.go:15:2: google.golang.org/grpc@v1.41.0: Get "https://proxy.golang.org/google.golang.org/grpc/@v/v1.41.0.zip": dial tcp 172.217.163.49:443: i/o timeout
../../transport/internet/grpc/encoding/conn.go:13:2: google.golang.org/grpc@v1.41.0: Get "https://proxy.golang.org/google.golang.org/grpc/@v/v1.41.0.zip": dial tcp 172.217.163.49:443: i/o timeout
../../app/commander/service.go:10:2: google.golang.org/grpc@v1.41.0: Get "https://proxy.golang.org/google.golang.org/grpc/@v/v1.41.0.zip": dial tcp 172.217.163.49:443: i/o timeout
../../app/log/command/config_grpc.pb.go:9:2: google.golang.org/grpc@v1.41.0: Get "https://proxy.golang.org/google.golang.org/grpc/@v/v1.41.0.zip": dial tcp 172.217.163.49:443: i/o timeout
[... cut ...]
make[2]: *** [package/Makefile:116: package/feeds/openmptcprouter/v2ray-core/compile] Error 1
make[2]: Leaving directory '/home/osboxes/openmptcprouter/imx6/source'
make[1]: *** [package/Makefile:110: /home/osboxes/openmptcprouter/imx6/source/staging_dir/target-arm_cortex-a9+neon_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/osboxes/openmptcprouter/imx6/source'
make: *** [/home/osboxes/openmptcprouter/imx6/source/include/toplevel.mk:230: world] Error 2

## Specifications

  - Debian VirtualBox version: <Debian 10>
  - OpenMPTCProuter platform: <imx6>

atyouspace, 2022-04-20:
Is there a way to replace the addresses “[https://proxy.golang.org]” in all '.go' files under package with the new?
Because I found if I use "wget https://proxy.golang.org/...", it is timeout. 
But "https://goproxy.cn" works for me which is a local proxy from official.
@Ysurac

Ysurac, 2022-04-20:
Golang is used by v2ray, use "make menuconfig" in source/<yourarch> to remove it.
You may also try to do a "export GOPROXY=https://goproxy.cn" before building, this should work.

atyouspace, 2022-04-24:
Thanks for @Ysurac . This issue has been solved.

closed 2022-04-24 by atyouspace