Ysurac/openmptcprouter #2337: ERR_CONNECTION_REFUSED with MPTCP over VPN

skmshd, 2022-04-23:
Configuration:
- Proxy: V2Ray
- VPN: MLVPN
- MPTCP over VPN: WireGuard

I have 4 WANs (3 cellular + 1 fiber), but WAN3 has "no ping" issue (confirmed by "ping -B -I WAN3 VPS"), so I turned on "MPTCP over VPN" with WireGuard. 
Immediately thereafter, the strange behavior begins. The chrome browser on my client started showing "ERR_CONNECTION_REFUSED" for all web accesses.

I tried to disable the interfaces (WAN3, wireGuardWAN3) in the interface UI, but it didn't help. Switching to Shadowsocks proxy nor OpenVPN also didn't help. Only the way I can fix this was disabling "MPTCP over VPN" on WAN3. 

First of all, I would like to know my understanding is correct. In case WAN blocked ICMP, I should use "MPTCP over VPN", am I right?\
My detailed setting is available [here](https://docs.google.com/document/d/1opJt5VChrZdfNPABVlEFu6cZPKIMP04t-HJwmxUtfa0/edit?usp=sharing) (used Google Drive because of max char limtation), so could anyone advise me?

Ysurac, 2022-04-24:
Please use the template.
V2Ray doesn't work with MPTCP over VPN.
You should use MPTCP over VPN only when MPTCP is blocked on a connection. When ICMP is blocked you only have to change the type of connection tracking in Service->OMR-Tracker.

skmshd, 2022-04-26:
Thank you so much, Ysurac, for your prompt support. I totally misunderstood that the feature of MPTCP over VPN. Actually OMR-Tracker configuration change resolved my issue. Thanks

closed 2022-04-26 by skmshd