Ysurac/openmptcprouter #2343: Additional help with OMR and pfSense

davidvermillion, 2022-05-03:
I'm trying to make pfSense work in front of OMR. I found this in the documentation:

```
On pfSense, add a WAN interface with OMR as the default gateway.
Disable Outbound NAT on pfSense.
On OMR, add static route(s) to your LAN network(s).
If you need port forwarding, redirect the needed ports to the target LAN hosts on OMR, then add a Pass WAN firewall rule on pfSense to allow inbound traffic to the target host and port.
```

Is there anyone who can give more information on how to do this?

- I've configured the LAN and WAN with the upstream gateway pointing to OMR (and internet works, its just double NATed) 
- As soon as I disable outbound NAT on pfSense, everything quits working, and I can't access LUCI on OMR
- I'm not sure what the static route on OMR is supposed to look like

[image]

[image]

davidvermillion, 2022-05-03:
@0xgrm It looks like you were the person that originally came up with this method. If you're still using this config, I'd appreciate any hints.

closed 2022-05-08 by davidvermillion

0xgrm, 2022-06-16:
I am not using OMR anymore, since I now have a 10 Gbps fiber connection. 😊

You must create a static route in OMR to your LAN network through your pfSense gateway.

If you don't do that OMR has no idea where to return the packets coming from your LAN network without NAT.