Ysurac/openmptcprouter #2344: Google locked my VPS due to "cryptocurrency mining"

tbazooka, 2022-05-04:
I've been using OMR for a few months now, and a few weeks ago I received an email from Google Cloud Platform stating that my VPS was been used for cryptocurrency mining and because that is against the ToS they suspended my VPS. At that time, I thought that the VPS may have been compromised (couldn't find any evidence of that) so I dropped all the project in Google Cloud and provisioned a brand new VPS, after a few weeks of working without any issue, Google sent me again the same warning, I've made an appeal asking for any evidence of cryptocurrency mining in the VPS but they just replied that my project was not allowed anymore to run in Google Cloud Platform. 

I'm pretty sure that the server was not been used for mining, for two reasons; I made an specific project with random credentials been used for this, and second, because the server didn't show any particular spikes in resources consumption.

I can only conclude that GCP is tagging some process on the VPS as cryptocurrency mining.

VPS: Debian 10 with master branch.

Ysurac, 2022-05-04:
If a client computer due some mining it can be seen by the VPS provider as VPS mining.
I didn't had any problem with GCP for now. But they may have a broken detect script.

sciro-de, 2022-05-16:
I am having the exact same problem. I got a mining right in my network using nicehash. Maybe they rate the connection to those mining servers as "mining"... which is, to put it blunt, plain retarded of google..... I made an appeal too and explained everything to them, but I just get an automatic answer. They don't even look into it. I will put my mining stuff on a separate VLAN now and bypass the MPTCP in my opnsense directly to one of my DSL connections, that should solve the issue.

closed 2022-08-19 by github-actions[bot]