Ysurac/openmptcprouter #2345: DNS_PROBE error

censorshipm, 2022-05-06:
## Current Behavior
Hello,
I'm using the beta version.
When I use the VPS server, I got `DNS_PROBE_STARTED` or `DNS_PROBE_FINISHED_NO_INTERNET` error for all websites.
It worked for like 2 weeks, and now it happens everytimes.
[image]

The server is reachable, and it happens even if I reinstall the router and the VPS. I have tested with two different VPS (Milkywan and Sapinet)

[image]

Here's my log:

```Fri May  6 21:10:12 2022 user.notice omr-bypass: Starting OMR-ByPass...
Fri May  6 21:10:13 2022 user.notice omr-bypass: OMR-ByPass is running
Fri May  6 21:10:15 2022 user.notice omr-tracker: Launching...
Fri May  6 21:10:21 2022 daemon.err miniupnpd[13191]: NLMSG_ERROR -13 : Permission denied
Fri May  6 21:10:21 2022 daemon.warn miniupnpd[13191]: SSDP packet sender [fd08:31a4:36c6:a000:5c19:282b:2df6:ebda]:32840 (if_index=-1) not from a LAN, ignoring
Fri May  6 21:10:21 2022 daemon.err miniupnpd[13191]: NLMSG_ERROR -13 : Permission denied
Fri May  6 21:10:21 2022 daemon.warn miniupnpd[13191]: SSDP packet sender [fd08:31a4:36c6:a000:5c19:282b:2df6:ebda]:32840 (if_index=-1) not from a LAN, ignoring
Fri May  6 21:10:24 2022 user.notice omr-tracker: Launched
Fri May  6 21:10:30 2022 daemon.err miniupnpd[13191]: NLMSG_ERROR -13 : Permission denied
Fri May  6 21:10:30 2022 daemon.warn miniupnpd[13191]: SSDP packet sender [fd08:31a4:36c6:a000:a03d:2467:12f5:41e9]:54947 (if_index=-1) not from a LAN, ignoring
Fri May  6 21:10:30 2022 daemon.err miniupnpd[13191]: NLMSG_ERROR -13 : Permission denied
Fri May  6 21:10:30 2022 daemon.warn miniupnpd[13191]: SSDP packet sender [fd08:31a4:36c6:a000:a03d:2467:12f5:41e9]:54947 (if_index=-1) not from a LAN, ignoring
Fri May  6 21:10:31 2022 daemon.err miniupnpd[13191]: NLMSG_ERROR -13 : Permission denied
Fri May  6 21:10:31 2022 daemon.warn miniupnpd[13191]: SSDP packet sender [fd08:31a4:36c6:a000:34d1:1386:cd08:8284]:36396 (if_index=-1) not from a LAN, ignoring
Fri May  6 21:10:31 2022 daemon.err miniupnpd[13191]: NLMSG_ERROR -13 : Permission denied
Fri May  6 21:10:31 2022 daemon.warn miniupnpd[13191]: SSDP packet sender [fd08:31a4:36c6:a000:34d1:1386:cd08:8284]:36396 (if_index=-1) not from a LAN, ignoring
Fri May  6 21:10:41 2022 daemon.err miniupnpd[13191]: NLMSG_ERROR -13 : Permission denied
Fri May  6 21:10:41 2022 daemon.warn miniupnpd[13191]: SSDP packet sender [fd08:31a4:36c6:a000:5c19:282b:2df6:ebda]:32840 (if_index=-1) not from a LAN, ignoring
Fri May  6 21:10:41 2022 daemon.err miniupnpd[13191]: NLMSG_ERROR -13 : Permission denied
Fri May  6 21:10:41 2022 daemon.warn miniupnpd[13191]: SSDP packet sender [fd08:31a4:36c6:a000:5c19:282b:2df6:ebda]:32840 (if_index=-1) not from a LAN, ignoring
Fri May  6 21:10:50 2022 daemon.err miniupnpd[13191]: NLMSG_ERROR -13 : Permission denied
Fri May  6 21:10:50 2022 daemon.warn miniupnpd[13191]: SSDP packet sender [fd08:31a4:36c6:a000:a03d:2467:12f5:41e9]:54947 (if_index=-1) not from a LAN, ignoring
Fri May  6 21:10:50 2022 daemon.err miniupnpd[13191]: NLMSG_ERROR -13 : Permission denied
Fri May  6 21:10:50 2022 daemon.warn miniupnpd[13191]: SSDP packet sender [fd08:31a4:36c6:a000:a03d:2467:12f5:41e9]:54947 (if_index=-1) not from a LAN, ignoring
Fri May  6 21:10:51 2022 daemon.err miniupnpd[13191]: NLMSG_ERROR -13 : Permission denied
Fri May  6 21:10:51 2022 daemon.warn miniupnpd[13191]: SSDP packet sender [fd08:31a4:36c6:a000:34d1:1386:cd08:8284]:36396 (if_index=-1) not from a LAN, ignoring
Fri May  6 21:10:51 2022 daemon.err miniupnpd[13191]: NLMSG_ERROR -13 : Permission denied
Fri May  6 21:10:51 2022 daemon.warn miniupnpd[13191]: SSDP packet sender [fd08:31a4:36c6:a000:34d1:1386:cd08:8284]:36396 (if_index=-1) not from a LAN, ignoring
Fri May  6 21:11:01 2022 daemon.err miniupnpd[13191]: NLMSG_ERROR -13 : Permission denied
Fri May  6 21:11:01 2022 daemon.warn miniupnpd[13191]: SSDP packet sender [fd08:31a4:36c6:a000:5c19:282b:2df6:ebda]:32840 (if_index=-1) not from a LAN, ignoring
Fri May  6 21:11:01 2022 daemon.err miniupnpd[13191]: NLMSG_ERROR -13 : Permission denied
Fri May  6 21:11:01 2022 daemon.warn miniupnpd[13191]: SSDP packet sender [fd08:31a4:36c6:a000:5c19:282b:2df6:ebda]:32840 (if_index=-1) not from a LAN, ignoring
Fri May  6 21:11:10 2022 daemon.err miniupnpd[13191]: NLMSG_ERROR -13 : Permission denied
Fri May  6 21:11:10 2022 daemon.warn miniupnpd[13191]: SSDP packet sender [fd08:31a4:36c6:a000:a03d:2467:12f5:41e9]:54947 (if_index=-1) not from a LAN, ignoring
Fri May  6 21:11:10 2022 daemon.err miniupnpd[13191]: NLMSG_ERROR -13 : Permission denied
Fri May  6 21:11:10 2022 daemon.warn miniupnpd[13191]: SSDP packet sender [fd08:31a4:36c6:a000:a03d:2467:12f5:41e9]:54947 (if_index=-1) not from a LAN, ignoring
Fri May  6 21:11:11 2022 daemon.err miniupnpd[13191]: NLMSG_ERROR -13 : Permission denied
Fri May  6 21:11:11 2022 daemon.warn miniupnpd[13191]: SSDP packet sender [fd08:31a4:36c6:a000:34d1:1386:cd08:8284]:36396 (if_index=-1) not from a LAN, ignoring
Fri May  6 21:11:11 2022 daemon.err miniupnpd[13191]: NLMSG_ERROR -13 : Permission denied
Fri May  6 21:11:11 2022 daemon.warn miniupnpd[13191]: SSDP packet sender [fd08:31a4:36c6:a000:34d1:1386:cd08:8284]:36396 (if_index=-1) not from a LAN, ignoring
Fri May  6 21:11:21 2022 daemon.err miniupnpd[13191]: NLMSG_ERROR -13 : Permission denied
Fri May  6 21:11:21 2022 daemon.warn miniupnpd[13191]: SSDP packet sender [fd08:31a4:36c6:a000:5c19:282b:2df6:ebda]:32840 (if_index=-1) not from a LAN, ignoring
Fri May  6 21:11:21 2022 daemon.err miniupnpd[13191]: NLMSG_ERROR -13 : Permission denied
Fri May  6 21:11:21 2022 daemon.warn miniupnpd[13191]: SSDP packet sender [fd08:31a4:36c6:a000:5c19:282b:2df6:ebda]:32840 (if_index=-1) not from a LAN, ignoring
Fri May  6 21:11:30 2022 daemon.err miniupnpd[13191]: NLMSG_ERROR -13 : Permission denied
Fri May  6 21:11:30 2022 daemon.warn miniupnpd[13191]: SSDP packet sender [fd08:31a4:36c6:a000:a03d:2467:12f5:41e9]:54947 (if_index=-1) not from a LAN, ignoring
Fri May  6 21:11:30 2022 daemon.err miniupnpd[13191]: NLMSG_ERROR -13 : Permission denied
Fri May  6 21:11:30 2022 daemon.warn miniupnpd[13191]: SSDP packet sender [fd08:31a4:36c6:a000:a03d:2467:12f5:41e9]:54947 (if_index=-1) not from a LAN, ignoring
Fri May  6 21:11:31 2022 daemon.err miniupnpd[13191]: NLMSG_ERROR -13 : Permission denied
Fri May  6 21:11:31 2022 daemon.warn miniupnpd[13191]: SSDP packet sender [fd08:31a4:36c6:a000:34d1:1386:cd08:8284]:36396 (if_index=-1) not from a LAN, ignoring
Fri May  6 21:11:31 2022 daemon.err miniupnpd[13191]: NLMSG_ERROR -13 : Permission denied
Fri May  6 21:11:31 2022 daemon.warn miniupnpd[13191]: SSDP packet sender [fd08:31a4:36c6:a000:34d1:1386:cd08:8284]:36396 (if_index=-1) not from a LAN, ignoring
Fri May  6 21:11:41 2022 daemon.err miniupnpd[13191]: NLMSG_ERROR -13 : Permission denied
Fri May  6 21:11:41 2022 daemon.warn miniupnpd[13191]: SSDP packet sender [fd08:31a4:36c6:a000:5c19:282b:2df6:ebda]:32840 (if_index=-1) not from a LAN, ignoring
Fri May  6 21:11:41 2022 daemon.err miniupnpd[13191]: NLMSG_ERROR -13 : Permission denied
Fri May  6 21:11:41 2022 daemon.warn miniupnpd[13191]: SSDP packet sender [fd08:31a4:36c6:a000:5c19:282b:2df6:ebda]:32840 (if_index=-1) not from a LAN, ignoring
```

## Specifications

  - OpenMPTCProuter version: v0.59beta5-5.15 r0+19136-83ca16fc43
  - OpenMPTCProuter VPS version: 0.1027-test 5.15.13-mptcp
  - OpenMPTCProuter VPS provider: Milkywan - Sapinet
  - OpenMPTCProuter platform: Raspberry Pi 4
  - Country: France

Ysurac, 2022-05-06:
Check ip, gateway and DNS of your client computer. As all seems to be working from the router.

censorshipm, 2022-05-06:
All my clients have the same problem. They all have automatic ip/dns/gateway. Also, internet works when I'm using the non-beta VPS script

Ysurac, 2022-05-06:
Can you do a "ping 1.1.1.1" from your client computer ? If yes, does it work if you set "1.1.1.1" as DNS.

censorshipm, 2022-05-06:
Ping 1.1.1.1 just get timeout

Ysurac, 2022-05-07:
Via ssh on the router, what is the result of ```ping -c 3 1.1.1.1```, ```curl 1.1.1.1``` and ```iptables-save | grep REDIRECT``` ?

closed 2022-08-10 by github-actions[bot]