Ysurac/openmptcprouter #2348: Precision on how to update OMR running in a ESXI VM

SLTech-SPRL, 2022-05-12:
## Expected Behavior

Update OMR running inside ESXI VM by using "System->Backup/Flash Firmware->Backup->Flash new firmware image"
I would like to update from [v0.58.5 r0+16336-b36068d35d] to [v0.59beta5-5.4-r0+16520-864bba55d8]
## Current Behavior

I read the doc but i'm not sure on which image to use to proceed this update.
The VM was built from the "openmptcprouter-v0.58.5-r0+16336-b36068d35d-x86-64-generic-ext4-combined" VDMK
The doc say "Download the image (.img) then use System->Backup/Flash Firmware->Backup->Flash new firmware image"

Must I use 
- openmptcprouter-v0.59beta5-5.4-r0+16520-864bba55d8-x86-64-generic-kernel.bin
or
- openmptcprouter-v0.59beta5-5.4-r0+16520-864bba55d8-x86-64-generic-ext4-combined.img

Or neither, because for other targets I see there are "sysupgrade" tagged images available but not for X86_64 targets ?

Sorry I know that it's for sure a newbie question but I'm not familiarized with this type of process.

Thanks
## Specifications

  - OpenMPTCProuter version: v0.58.5 r0+16336-b36068d35d
  - OpenMPTCProuter VPS version: 0.1026 5.4.100-mptcp
  - OpenMPTCProuter VPS provider: OVH/PulseHeberg
  - OpenMPTCProuter platform: ESXI VM

Ysurac, 2022-05-12:
As indicated in the doc, you need to use the .img image, so "openmptcprouter-v0.59beta5-5.4-r0+16520-864bba55d8-x86-64-generic-ext4-combined.img"
Remember that v0.59beta5 is still a beta. There is some bugs ;)

Ysurac, 2022-05-12:
"*kernel.bin" is only the kernel image.

closed 2022-08-16 by github-actions[bot]