Ysurac/openmptcprouter #2349: Recommended USB 3.0 to Ethernet Adapter for RPI4

slazinger7, 2022-05-13:
Our OpenMPTCProuter setup was previously flaky running v0.57.3 (e.g., on restart or power loss, the router was unable to connect to the gateways, connect to the VPS, etc.).  As long as the system didn't go down, everything worked fine.

After upgrading to (and trying a fresh install of) v0.58.5, the gateways consistently go down (and in some cases completely, where we need to reload the firmware) and it appears to be a result of errors with the USB adapter (the modems and internet still work fine).  I suspect the flaky behavior on v57.3 was because of the adapter.

FYI - we have two of these adapters:

https://www.newegg.com/p/36F-00MF-00001?Item=9SIAPY9CJC9282

A forum I was reading on our recent disconnection issue on v0.58.5 (which I unfortunately can't find the link to now) indicated that this type of "generic" USB to Ethernet adapter was problematic for RPI4 - the forum member was experiencing the same issue we are seeing in the logs.

Question - can anyone make any recommendations for a USB 3.0 to gigabit Ethernet adapter that would be particularly stable for use with OpenMPTCProuter?

I'd prefer to get different (and known to be stable) hardware, rather than try to troubleshoot the existing devices.

It would be used with the latest version:
    OpenMPTCProuter version: v0.58.5 r0+16336-b36068d35d
    OpenMPTCProuter VPS version: Ubuntu
    OpenMPTCProuter VPS provider: Self
    OpenMPTCProuter platform: RPI4B

TalalMash, 2022-05-15:
(AX88179)[https://deviwiki.com/wiki/Wired_network_adapters] based dongles are historically free of issues.

islander261, 2022-05-18:
My experience is the exact opposite. I found Realtek chip base dongles work much better. TP-Link UE-300 (Realtek based), two of them work perfectly. The TP-306 (Axis based), didn't work reliably at all, random hangs every few days that required a reboot to fix.

TalalMash, 2022-05-18:
Accidentally removed post, stable EEPROM for ASIX AX88179 with in-tree kernel module, write using ethtool -e:

```
00 50 b6 83 60 a3 90 17 95 0b b7 73 00 e0 3e 01
80 0b 09 04 0e 07 1a 10 26 0e 2d 16 41 58 38 38
31 37 39 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 41 53 49 58 20 45 6c 65 63 2e 20 43
6f 72 70 2e 00 00 00 00 00 00 00 00 30 30 30 30
35 30 42 36 38 33 36 30 3a 33 05 0f 16 00 02 07
10 02 02 00 00 00 0a 10 03 00 0e 00 01 0a ff 07
00 00 00 00 00 00 00 00 f3 ff 40 4a 40 00 40 30
0d 49 90 41 c1 bd ff ff ff ff ff ff ff ff ff ff
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
```

Stable RTL8153 EEPROM with 8152 kmod: 
Plug to a Windows machine and download latest driver from Realtek, driver autoupdate EEPROM parameters.

slazinger7, 2022-06-17:
To post a potential solution for others, we ended up buying a pair of the Dell USB 3.0 to Ethernet PXE Boot adapters, shown here:

https://www.dell.com/en-us/shop/dell-adapter-usb-30-to-ethernet-pxe-boot/apd/443-bbbd/pc-accessories

The issues seems to have been resolved.

We still get a fair number of the following errors:

daemon.err /usr/bin/ss-local[4021]: getpeername: Socket not connected

But, my understanding is these aren't that uncommon.  That said, the internet seems to be working just fine (no disconnects), and speeds are good (usually between 150-200 Mbps down).

closed 2022-09-21 by github-actions[bot]