Ysurac/openmptcprouter #2352: Upgrading Server to beta

myshophosting, 2022-05-17:
Hi

I was able to successfully update my router (x86) Via System -> Sysupgrade - this went smoothly

however the VPS I'm not sure how to upgrade. Does it require an update or just the router?

Ysurac, 2022-05-17:
Via System->Sysupgrade only stable release are available.
VPS need to be updated after each major version (0.54 to 0.55) or when it's indicated in the message on openmptcprouter.com

myshophosting, 2022-05-17:
Hi!

Im at v0.59beta6-5.4 r0+16591-b4ea8e1089 - on my router.

However im not clear about the VPS - does it need updating ? I updated my router from v0.58.5 to the above, and am not sure if the VPS server at the DC needs a corresponding update or not?

myshophosting, 2022-05-17:
on the dashboard it says: Version：5.4.100-mptcp 0.1026

myshophosting, 2022-05-17:
Ahh - just found this: wget -O - http://www.openmptcprouter.com/server-test/debian10-x86_64.sh | sh
I will run this.

myshophosting, 2022-05-17:
All fixed and udpated.

closed 2022-05-17 by myshophosting