Ysurac/openmptcprouter #2355: kernal does not update to mptcp (vps from buyvm) Debian 10 KVM Linux localhost 4.19.0-18-amd64

hle5128, 2022-05-20:
after finish the scrip: 
After reboot, check with uname -a that the kernel name contain mptcp.
 Else, you may have to modify GRUB_DEFAULT in /etc/default/grub

check uname-a
root@localhost:~# uname -a
Linux localhost 4.19.0-18-amd64 #1 SMP Debian 4.19.208-1 (2021-09-29) x86_64 GNU/Linux

I dont understand, i have 3 other hosts and they are all updated to mptcp just fine. 
What should i do ?

Here is my current /boot/grub/grub.cfg file

`#
# DO NOT EDIT THIS FILE
#
# It is automatically generated by grub-mkconfig using templates
# from /etc/grub.d and settings from /etc/default/grub
#

### BEGIN /etc/grub.d/00_header ###
if [ -s $prefix/grubenv ]; then
  set have_grubenv=true
  load_env
fi
if [ "${next_entry}" ] ; then
   set default="${next_entry}"
   set next_entry=
   save_env next_entry
   set boot_once=true
else
   set default="0"
fi

if [ x"${feature_menuentry_id}" = xy ]; then
  menuentry_id_option="--id"
else
  menuentry_id_option=""
fi

export menuentry_id_option

if [ "${prev_saved_entry}" ]; then
  set saved_entry="${prev_saved_entry}"
  save_env saved_entry
  set prev_saved_entry=
  save_env prev_saved_entry
  set boot_once=true
fi

function savedefault {
  if [ -z "${boot_once}" ]; then
    saved_entry="${chosen}"
    save_env saved_entry
  fi
}
function load_video {
  if [ x$feature_all_video_module = xy ]; then
    insmod all_video
  else
    insmod efi_gop
    insmod efi_uga
    insmod ieee1275_fb
    insmod vbe
    insmod vga
    insmod video_bochs
    insmod video_cirrus
  fi
}

if [ x$feature_default_font_path = xy ] ; then
   font=unicode
else
insmod part_msdos
insmod ext2
if [ x$feature_platform_search_hint = xy ]; then
  search --no-floppy --fs-uuid --set=root  dd752d20-4fc9-4bca-9ef7-b98ac7cf7d6f
else
  search --no-floppy --fs-uuid --set=root dd752d20-4fc9-4bca-9ef7-b98ac7cf7d6f
fi
    font="/usr/share/grub/unicode.pf2"
fi

if loadfont $font ; then
  set gfxmode=auto
  load_video
  insmod gfxterm
  set locale_dir=$prefix/locale
  set lang=en_US
  insmod gettext
fi
terminal_output gfxterm
if [ "${recordfail}" = 1 ] ; then
  set timeout=30
else
  if [ x$feature_timeout_style = xy ] ; then
    set timeout_style=menu
    set timeout=5
  # Fallback normal timeout code in case the timeout_style feature is
  # unavailable.
  else
    set timeout=5
  fi
fi
### END /etc/grub.d/00_header ###

### BEGIN /etc/grub.d/05_debian_theme ###
set menu_color_normal=cyan/blue
set menu_color_highlight=white/blue
### END /etc/grub.d/05_debian_theme ###

### BEGIN /etc/grub.d/10_linux ###
function gfxmode {
        set gfxpayload="${1}"
}
set linux_gfx_mode=
export linux_gfx_mode
menuentry 'Debian GNU/Linux' --class debian --class gnu-linux --class gnu --class os $menuentry_id_option 'gnulinux-simple-dd752d20-4fc9-4bca-9ef7-b98ac7cf7d6f' {
        load_video
        insmod gzio
        if [ x$grub_platform = xxen ]; then insmod xzio; insmod lzopio; fi
        insmod part_msdos
        insmod ext2
        if [ x$feature_platform_search_hint = xy ]; then
          search --no-floppy --fs-uuid --set=root  dd752d20-4fc9-4bca-9ef7-b98ac7cf7d6f
        else
          search --no-floppy --fs-uuid --set=root dd752d20-4fc9-4bca-9ef7-b98ac7cf7d6f
        fi
        echo    'Loading Linux 4.19.0-18-amd64 ...'
        linux   /boot/vmlinuz-4.19.0-18-amd64 root=UUID=dd752d20-4fc9-4bca-9ef7-b98ac7cf7d6f ro net.ifnames=0 biosdevname=0 debian-installer=en_US.UTF-8 net.ifnames=0 biosdevname=0 net.ifnames=0 biosdevname=0
        echo    'Loading initial ramdisk ...'
        initrd  /boot/initrd.img-4.19.0-18-amd64
}
submenu 'Advanced options for Debian GNU/Linux' $menuentry_id_option 'gnulinux-advanced-dd752d20-4fc9-4bca-9ef7-b98ac7cf7d6f' {
        menuentry 'Debian GNU/Linux, with Linux 4.19.0-18-amd64' --class debian --class gnu-linux --class gnu --class os $menuentry_id_option 'gnulinux-4.19.0-18-amd64-advanced-dd752d20-4fc9-4bca-9ef7-b98ac7cf7d6f' {
                load_video
                insmod gzio
                if [ x$grub_platform = xxen ]; then insmod xzio; insmod lzopio; fi
                insmod part_msdos
                insmod ext2
                if [ x$feature_platform_search_hint = xy ]; then
                  search --no-floppy --fs-uuid --set=root  dd752d20-4fc9-4bca-9ef7-b98ac7cf7d6f
                else
                  search --no-floppy --fs-uuid --set=root dd752d20-4fc9-4bca-9ef7-b98ac7cf7d6f
                fi
                echo    'Loading Linux 4.19.0-18-amd64 ...'
                linux   /boot/vmlinuz-4.19.0-18-amd64 root=UUID=dd752d20-4fc9-4bca-9ef7-b98ac7cf7d6f ro net.ifnames=0 biosdevname=0 debian-installer=en_US.UTF-8 net.ifnames=0 biosdevname=0 net.ifnames=0 biosdevname=0
                echo    'Loading initial ramdisk ...'
                initrd  /boot/initrd.img-4.19.0-18-amd64
        }
        menuentry 'Debian GNU/Linux, with Linux 4.19.0-18-amd64 (recovery mode)' --class debian --class gnu-linux --class gnu --class os $menuentry_id_option 'gnulinux-4.19.0-18-amd64-recovery-dd752d20-4fc9-4bca-9ef7-b98ac7cf7d6f' {
                load_video
                insmod gzio
                if [ x$grub_platform = xxen ]; then insmod xzio; insmod lzopio; fi
                insmod part_msdos
                insmod ext2
                if [ x$feature_platform_search_hint = xy ]; then
                  search --no-floppy --fs-uuid --set=root  dd752d20-4fc9-4bca-9ef7-b98ac7cf7d6f
                else
                  search --no-floppy --fs-uuid --set=root dd752d20-4fc9-4bca-9ef7-b98ac7cf7d6f
                fi
                echo    'Loading Linux 4.19.0-18-amd64 ...'
                linux   /boot/vmlinuz-4.19.0-18-amd64 root=UUID=dd752d20-4fc9-4bca-9ef7-b98ac7cf7d6f ro single net.ifnames=0 biosdevname=0 debian-installer=en_US.UTF-8 net.ifnames=0 biosdevname=0
                echo    'Loading initial ramdisk ...'
                initrd  /boot/initrd.img-4.19.0-18-amd64
        }
        menuentry 'Debian GNU/Linux, with Linux 4.19.0-14-amd64' --class debian --class gnu-linux --class gnu --class os $menuentry_id_option 'gnulinux-4.19.0-14-amd64-advanced-dd752d20-4fc9-4bca-9ef7-b98ac7cf7d6f' {
                load_video
                insmod gzio
                if [ x$grub_platform = xxen ]; then insmod xzio; insmod lzopio; fi
                insmod part_msdos
                insmod ext2
                if [ x$feature_platform_search_hint = xy ]; then
                  search --no-floppy --fs-uuid --set=root  dd752d20-4fc9-4bca-9ef7-b98ac7cf7d6f
                else
                  search --no-floppy --fs-uuid --set=root dd752d20-4fc9-4bca-9ef7-b98ac7cf7d6f
                fi
                echo    'Loading Linux 4.19.0-14-amd64 ...'
                linux   /boot/vmlinuz-4.19.0-14-amd64 root=UUID=dd752d20-4fc9-4bca-9ef7-b98ac7cf7d6f ro net.ifnames=0 biosdevname=0 debian-installer=en_US.UTF-8 net.ifnames=0 biosdevname=0 net.ifnames=0 biosdevname=0
                echo    'Loading initial ramdisk ...'
                initrd  /boot/initrd.img-4.19.0-14-amd64
        }
        menuentry 'Debian GNU/Linux, with Linux 4.19.0-14-amd64 (recovery mode)' --class debian --class gnu-linux --class gnu --class os $menuentry_id_option 'gnulinux-4.19.0-14-amd64-recovery-dd752d20-4fc9-4bca-9ef7-b98ac7cf7d6f' {
                load_video
                insmod gzio
                if [ x$grub_platform = xxen ]; then insmod xzio; insmod lzopio; fi
                insmod part_msdos
                insmod ext2
                if [ x$feature_platform_search_hint = xy ]; then
                  search --no-floppy --fs-uuid --set=root  dd752d20-4fc9-4bca-9ef7-b98ac7cf7d6f
                else
                  search --no-floppy --fs-uuid --set=root dd752d20-4fc9-4bca-9ef7-b98ac7cf7d6f
                fi
                echo    'Loading Linux 4.19.0-14-amd64 ...'
                linux   /boot/vmlinuz-4.19.0-14-amd64 root=UUID=dd752d20-4fc9-4bca-9ef7-b98ac7cf7d6f ro single net.ifnames=0 biosdevname=0 debian-installer=en_US.UTF-8 net.ifnames=0 biosdevname=0
                echo    'Loading initial ramdisk ...'
                initrd  /boot/initrd.img-4.19.0-14-amd64
        }
}

### END /etc/grub.d/10_linux ###

### BEGIN /etc/grub.d/20_linux_xen ###

### END /etc/grub.d/20_linux_xen ###

### BEGIN /etc/grub.d/30_os-prober ###
### END /etc/grub.d/30_os-prober ###

### BEGIN /etc/grub.d/30_uefi-firmware ###
### END /etc/grub.d/30_uefi-firmware ###

### BEGIN /etc/grub.d/40_custom ###
# This file provides an easy way to add custom menu entries.  Simply type the
# menu entries you want to add after this comment.  Be careful not to change
# the 'exec tail' line above.
### END /etc/grub.d/40_custom ###

### BEGIN /etc/grub.d/41_custom ###
if [ -f  ${config_directory}/custom.cfg ]; then
  source ${config_directory}/custom.cfg
elif [ -z "${config_directory}" -a -f  $prefix/custom.cfg ]; then
  source $prefix/custom.cfg;
fi
### END /etc/grub.d/41_custom ###`

Ysurac, 2022-05-20:
Please use the template.
What is the VPS provider ?

Ysurac, 2022-05-20:
I see buyvm in the subject.
You are the 2nd user with same problem on this provider...
What is the package and what city ? I will buy one.

hle5128, 2022-05-20:
https://buyvm.net <https://buyvm.net/>
from your recommendation list.

hle5128, 2022-05-20:
> I see buyvm in the subject.
> You are the 2nd user with same problem on this provider...
> What is the package and what city ? I will buy one.

its New York - AMD RYZEN KVM - NY RYZEN KVM 1GB
I can DM you the access to the host to test.

hle5128, 2022-05-20:
> I see buyvm in the subject.
> You are the 2nd user with same problem on this provider...
> What is the package and what city ? I will buy one.

i sent info to your email ycarus@zugaina.org

closed 2022-08-24 by github-actions[bot]