Ysurac/openmptcprouter #2356: MacOS 12 does not pickup DHCP responses

Safarte-infra, 2022-05-21:
## Expected Behavior
When connecting a MacOS 12 device over wifi (MacBook), DHCP from Dnsmasq should provide an IP and let the device accessing internet.

## Current Behavior
Currently on Mac Devices, MacOSX 12 clients fails to get DHCP responses over Ethernet or Wifi.

## Possible Solution
None so far.

## Steps to Reproduce the Problem

  1. Install OpenmpTCProuter and connect MacOSX 12 device to it over ethernet or Wifi.
  2. Use network auto-configuration (DHCP client) in the network settings of the MacOSX settings.
  3. The Apple device never connects (not getting an IP from DHCP server). All other devices are connecting successfully.

## Context (Environment)
One x64 box with openmptcprouter last version as of today with single Ethernet interface.
One Freebox (French ISP provider) with DHCP disabled and IPv6 disabled.
One 5G Router with IPv6 disabled and DHCP disabled.
Freebox built-in switch is used.

## Specifications

  - OpenMPTCProuter version: v0.58.5-r0+16336-b36068d35d-x86-64
  - OpenMPTCProuter VPS version: v0.58.5
  - OpenMPTCProuter VPS provider: AlpineDC Lausanne, Swizerland (KVM).
  - OpenMPTCProuter platform: x86_64
  - Country: France

vinceducat, 2022-06-21:
@Safarte-infra  i have the same problem did you slove it ?

Safarte-infra, 2022-06-21:
Hello,

Yes, apparently, the ARP protocol get tricked due to the multiwan way of
working on the same broadcast domain. MacOs detects it as IP conflict and
it ends the lease. (I saw all this on Wireshark).
To work around this issue, I had to physically separate wans with dedicated
USB Ethernet adapters to separate broadcast domains. (15 bucks each).
After all, this architecture is more efficient and simple since I can keep
DHCPv4 and IPv6 enabled on my 2 wan interfaces and routers so it requires
less configuration and potentially I can enable IPv6 on MPTCPRouter later.

Hope this can help.

Safarte-infra, 2022-06-21:
> @Safarte-infra <https://github.com/Safarte-infra> so do you have 3
> ethernet pysical card on omr? 1 for l'an and 2 for wan 1 each?
> @Ysurac <https://github.com/Ysurac> any update on the Omr side il
> multiwan can souvent the pb

Yes, I have 3 physical interfaces, one lan and 2 wan.

vinceducat, 2022-06-21:
OK i have the samedi conf as you. Freebox ang 5g routeur freemobile. What is your hardware conf for OMR?

Safarte-infra, 2022-06-21:
> OK i have the same conf as you. Freebox ang 5g routeur freemobile. What is

OMR is a Celeron x64 small computer with a single 1Gb integrated interface.
Added 2 x USB 100 Mbps Ethernet adapter.
Then I configured eth1 (wan1) and eth2 (wan2) as PHYSICAL adapter in OMR
using DHCP.
It works like a charm.

Safarte-infra, 2022-06-21:
> i have a small computer with 1 phys card for lan and wan2 (5G) and 1 phys
> card for freebox V5 ... because Freebox must be on separate card due to the
> fact on freebox V5 i cant disable ipv6 and V5 dont be able to setup next op
> in ip v6.
>
> for me usb eth is not a good solution because in this case my speed (maxi
> 300Mbit/s) would fall down to 100Mbit/s du to speed limitation of usb..

Take a 1GB USB Ethernet adapter, it runs at least at 400Mbps if you have
usb2, if usb3 then you will be at 1 GB.

vinceducat, 2022-06-21:
ok i will try ;-)

Safarte-infra, 2022-06-21:
You can disable IPv6 on Freebox, just check openmptcp documentation ;-)

vinceducat, 2022-06-21:
will try...

vinceducat, 2022-06-22:
thanks for the solution @Safarte-infra 

ipv6 freebox doc is for FB revolution and over, my FB is v5 so lower

closed 2022-09-25 by github-actions[bot]