Ysurac/openmptcprouter #2358: sh: 301: grub-mkconfig: not found error

brackerp, 2022-05-23:
## Expected Behavior
Install MPTCP on fresh Debin 10 with no error 

## Current Behavior
Receved error "sh: 301: grub-mkconfig: not found"

## Possible Solution

## Steps to Reproduce the Problem

  1.Old VM at BuyVM reinstal Debin 10
  2.Start MPTCP script
  3.recieve error sh: 301: grub-mkconfig: not found error

## Context (Environment)
reintsall to see if repairs pihole install issue

## Specifications

  - OpenMPTCProuter version: 0.58.5
  - OpenMPTCProuter VPS version: Debian 10
  - OpenMPTCProuter VPS provider: BuyVm
  - OpenMPTCProuter platform: x86_64
  - Country: <USA

Ysurac, 2022-05-23:
Should be fixed now. I made a typo when trying to fix strange BuyVm VPS installation...

brackerp, 2022-05-23:
reninstalled debin 10
ran mptcp script
rebooted
no ssh at 65222 or 22

Ysurac, 2022-05-24:
What do you have on console/KVM view from BuyVM ?

brackerp, 2022-05-25:
At this time i have reinstalled in both buyvm and digitalocean and it installed correctly

brackerp, 2022-05-25:
I was wrong 
tomeny vms open
digitalocean works right it has a static ip in interface config
buyvm uses dhcp the ip is lost apfter the script

this is before
root@localhost:~# hostname -I
205.185.122.203 10.255.254.1 10.255.251.1 10.255.255.1 64:ff9b::aff:fb01
root@localhost:~#
[image]
this is after

this a log fo the install

[logfileBvm220524.txt](https://github.com/Ysurac/openmptcprouter/files/8768038/logfileBvm220524.txt)

Ysurac, 2022-05-25:
Strange, the firewall, shorewall, shouldn't block DHCP.
What is the interface name that have the public IP by default and what do you have in ```/etc/shorewall/params.net``` after installation ?

brackerp, 2022-05-25:
> On Wed, May 25, 2022 at 12:47 AM Ycarus (Yannick Chabanois) <

the picture has the vnc ip and password
mptcp is the root pw

i got to go to work

brackerp, 2022-05-25:
also i has installed the script in to a ubuntu  on this vm and it worked
fine

Ysurac, 2022-05-26:
I don't have the VNC password and it should be send via mail (contact@openmptcprouter.com) and not in public ;)
I think buyvm ubuntu install is correct with grub package installed, this doesn't seems to be the case on debian10/11 now.
VPS are out of stock, so I can't do any real tests.

brackerp, 2022-05-26:
in the png file i sent you in the previous email is the ip address and port
number and the password
i still have the admin console pw and will rebuild the vm once we are done
here is the ip out of the png file 209.141.63.46:10756
and pw AyrflXgh

Ysurac, 2022-05-26:
I fixed the issue on your VPS, the script is not able to detect default interface.
What is the result of "ip -o -4 route show to default" before running the install script ?

brackerp, 2022-05-26:
***@***.***:~# ip -o -4 route show to default
default via 205.185.122.1 dev eth0
***@***.***:~#

brackerp, 2022-05-26:
i reinstalled debian on vm
i have not run script yet

the vnc is still valid and password is valid

Ysurac, 2022-05-26:
ok, after reboot the default interface is "ens3", there is options "net.ifnames=0 biosdevname=0" in default /boot/grub/grub.cfg that I don't keep as I don't have default grub config files...
I will modify script but BuyVM are really doing a dirty install...

Ysurac, 2022-05-26:
This time both release and beta scripts should work with BuyVM.

brackerp, 2022-05-26:
looks like you got the script to install correctly

closed 2022-08-29 by github-actions[bot]