Ysurac/openmptcprouter #2359: ModemManager QMI 5G/LTE TX Packets No RX

Network-Traditions, 2022-05-23:
## Expected Behavior
Adding a Telit FN980 modem in QMI mode installed in a USB 3.0 sled as a ModemManager wan interface.  Upon saving and applying the configuration, the modem is added as configured, makes a connection with the T-Mobile network consistent with its SIM card provisioning and can successfully be utilized as a wan interface.

## Current Behavior
Upon saving and applying the configuration, the modem is added as configured, makes a connection with the T-Mobile network consistent with its SIM card provisioning and begins transmitting packets, however zero packets are received.  Pinging 8.8.8.8 freezes on first attempt.  It seems consistent with a firewall or routing issue.

Utilizing the same hardware as a QMI Cellular wan interface makes a connection with the T-Mobile network consistent with its SIM card provisioning and adds a DHCP IPV4 interface connected to the separate wwan0 interface and successfully transmits and receives packets.  Pinging 8.8.8.8 is successful as is omr-test-speed, however, it does not appear the modem is configured for carrier aggregation and delivers very limited bandwidth.  Setting up the interface as a sole master of the multipath connection fails to establish the connection.

Power cycling or rebooting the QMI Cellular configuration results in a change of the interface configuration where qmimux0 is muxed to wwan0.  Just like the ModemManager configuration, makes a connection with the T-Mobile network consistent with its SIM card provisioning and begins transmitting packets, however zero packets are received.

Similar behavior resulted when setting the Telit Fn980 to MBIM mode.  Implementing a QMI mode manual qmicli configuration from the command line also has a consistent result.  The interface appears to be properly configured and online, but it is unable to receive packets.  Coming from a pfSense experience discipline there is much I do not fully understand about OpenWRT and the OpenMPTCProuter.  When utilizing the T-Mobile service with their supplied router and connecting OpenMPTCProuter via a NATed Ethernet connection, utilizing the connection as a master or member of multipath is successful.  I've tested numerous variations to develop an better understanding of OpenWRT and OpenMPTCProuter.  During that testing for about 50% of the configurations, the Telit FN980 with T-mobile service has successfully passed the MPTCP Support Check.  The other 50% simple resulted in a timeout result. Additionally, at one point during a power cycle reboot, the modem was reconfigured as a ppp0 interface.  In this configuration, everything worked appropriately even as a sole master member of MPTCP, however, the connection was extremely slow.  Ultimately, a ModemManager wan interface with the Telit FN980 in QMI mode appears to be the best possible connection and most consistently maintained across power cycles and reboots if the RX packet reception issue can be resolved.  While T-Mobile does not echo pings against their gateways, this does not impact pinging the public internet (ie 8.8.8.8).  It seems the RX packet reception is a firewall or routing issue beyond my new experience with this platform, please feel free to request log information, additional test results, etc.  Thanks!

## Specifications

  - OpenMPTCProuter version:  v0.58.5 r0+16336-b36068d35d 
  - OpenMPTCProuter VPS version: 0.1026
  -     Virtualization: vmware
  Operating System: Debian GNU/Linux 10 (buster)
            Kernel: Linux 5.4.100-mptcp
      Architecture: x86-64

  - OpenMPTCProuter VPS provider: IONOS.com 2 vCore 2GB RAM "Type M VPS"
  - OpenMPTCProuter platform: x86_64 (HUNSN FNR-RS34G https://www.hunsn.com/item/network-security-firewall/pfsense-mini-pc)
  - Telit FN980 5G Modem (HWv1.0 https://www.telit.com/devices/fn980-and-fn980m-data-cards-support-5g/)
  - USB3.0 M.2 Key B Modem Adapter Enclosure (https://thewirelesshaven.com/shop/mini-pcie-m2-adapters/modem-enclosure/usb3-0-to-ngff-m-2-key-b-4g-5g-modem-adapter-enclosure-with-sim-card-slot-new-style/)

Ysurac, 2022-05-23:
ModemManager should always be used and modem in QMI mode is often also the better choice.
Make sure you configure interface to not set route and DNS.
You can also test latest beta release.

Network-Traditions, 2022-05-26:
Ysurac, thank you for your quick response and the openmptcprouter project!  The delay in my response has been due to extensive testing I've been conducting to both obtain a better working knowledge of this platform as well as to develop data that hopefully leads to the successful deployment of my configuration.

OpenWRT 22.03.0-rc-2 was deployed on the configuration to investigate the latest MODEMMANGER's performance.  This test was successful delivering +500Mbps download and +50Mbps upload connecting to a nearby T-Mobile 5gUC tower.  Subsequently, openMPTCProuter v0.59beta6-5.4 delivered similar successful results in a location with a far more degraded 5g only signal so the bandwidth is much less, but properly functioning.  The MODEMMANGER interface "Advanced Settings" tab provides checkboxes to disable the default gateway and DNS as suggested.  Additionally, copying the placeholder wan1 and wan2 settings of the out-of-the-box install of openMPTCProuter, the "Override IPv4 router table was set to wan" although I don't understand the purpose of this setting at this time.

Subsequently, I will continue testing v0.59 as it evolves and provide feedback where it will be helpful to the maturation of the project.  In closing, while the MODEMMANGER significantly established a stable and functioning wan interface, the Intel 225V revision 3 2.5GB NICs with the kmod-igc driver is less stable.  The observed instability to date has been an up down looping under certain conditions the specifics of which are unknown at  this time.  When reconnecting my StarLink v2 configured as a bridge mode DHCP eth1 connection member of MPTCP, the DHCP ip address assignment connects then disconnects every few seconds if the interface is brought up by anything other than the openMPTCProuter settings wizard.  Using the settings wizard, the interface is brought up, obtains its DHCP ip address and begins to function nominally.

closed 2022-05-26 by Network-Traditions