Ysurac/openmptcprouter #2363: OMR x live streaming SRT & VmixCall - Best UDP configuration and firewall management

simonemessina92, 2022-05-26:
Hi everyone, I've deployed 2 OMR on Linksys WRT 3200 ACM, running my 2 vps on OVH. Basically my main goal is to have a bonding router able to stream SRT ( https://www.haivision.com/resources/white-paper/srt-open-source/ ) from my vmix pc workstation (connected to OMR) to a remote location, and beeing able to see and get good quality videocalls with program such has Skype,zoom,teams and most importantly vmix call ( firewall option x vmixcall: https://www.vmix.com/knowledgebase/article.aspx/125/vmix-call-firewall-requirements ).

I've tried many different way in order to see what suite best my needs, I've tried both shadowsocks as proxy and v2ray, tried both glorytun TCP and glorytun UDP, tried all the MPTCP's congestion algorithm.

What actually happen is the following, with shadowsocks selected as proxy, and glorytun UDP selected as VPN, SRT work in listener mode without any problems, so i can forward the ip address of my OMR vps , the port I use for that stream and the REMI studio can catch my signal, but I've very bad problems with vmix call, bad quality disconnections and so on.

I've tried than to use glorytun TCP as VPN, still with shadowsocks as proxy, calls seems to work better, still not at maximum quality than. What I'm trying to understand is how to correctly setup the OMR firewall to have both the 2 things working , without having problems. From what I've been able to see glorytun TCP differentiate to glorytun UDP in the way the UDP traffic is sent/recieved, when I use glorytun TCP it seems only one connection at the time is beeing used, and the others work as seamless failover, when glorytun UDP is settled, I see all the connection load the traffic between themselves... I will attach some screenshoot I've done, if someone can help me to figured it out this I will much appreciated.

I've followed the method of OMR get 1 eth cable from a managed switch with vlan 802.1q configured, so 7 lte routers connected to the switch, than 1 trunk connected to OMR and I've create 7 diffrent wan, each with it's own vlan pivd configured, and static Ip address.  The normal behaviour of the video call with standard internet connection rather than using OMR, is 1280x720p 30 fps, but with OMR I've got many issues, pixelated image and really bad quality, I suppose it's someting totally firewall related, if some one can help me with the right configuration it will be much appreciated. This system is great, just need to understand how to make it work the way I need. Much appreciated any suggestion, thank you!

  - OpenMPTCProuter version: 0.59 beta6
  - OpenMPTCProuter VPS version: 0.59 beta6
  - OpenMPTCProuter VPS provider: OVH
  - OpenMPTCProuter platform:  LINKSYS WRT 3200 ACM
[image]
[image]
[image]
[image]
![170447927-58e5af7d-9d2b-4ef9-8748-d4430a403cea](htt
[image]
[image]
ps://user-images.githubusercontent.com/106303869/170590387-6d2e0646-6da8-45f5-8db5-a87b673d7b7e.png)

simonemessina92, 2022-05-28:
Update: the only way I was able to get the vmix call with good quality and stable without drops is using v2ray. From what I had understood, with v2ray both tcp and udp pass through the proxy, infact I've notice I can also put nothing in the vpn and vmix call still work. Now the thing that I miss is how to use SRT in listener mode, I mean how do I have to port forward a local ip machine in the LAN in order to be exposed over the public Ip address of the vps ? because if I use the forward menu in the firewall, even with vpn enabled, and use the rule I've post in the image here in the post it doesn't work so far... 

[image]

closed 2022-08-31 by github-actions[bot]

marccolemont, 2022-10-16:
SRT listener mode is fine for fixed installations. For events and locations every other day, I use a Haivision AWS SRT Gateway. That way all my encoders / decoders are in Caller mode. No Port forwarding needed on-site.