Ysurac/openmptcprouter #2369: cert issue installing VPS

packlet, 2022-05-29:
## Expected Behavior

trying to install VPS on stock Ubuntu 20.04 bare metal x86

install should work

## Current Behavior

script fails with cert errors

## Specifications

  - OpenMPTCProuter version: 
  - OpenMPTCProuter VPS version: latest release and latest snapshot
  - OpenMPTCProuter VPS provider: local PC 
  - OpenMPTCProuter platform: x86 bare metal

root@OMR-VPS:~# wget -O - https://www.openmptcprouter.com/server/ubuntu20.04-x86_64.sh | sh
...
Ign:28 https://repo.openmptcprouter.com buster/dists all c-n-f Metadata
Ign:3 https://repo.openmptcprouter.com buster/main all Packages
Err:4 https://repo.openmptcprouter.com buster/main amd64 Packages
  Certificate verification failed: The certificate is NOT trusted. The certificate chain uses expired certificate.  Could not handshake: Error in the certificate verification. [IP: 68.70.205.3 443]
Ign:6 https://repo.openmptcprouter.com buster/main Translation-en
Ign:7 https://repo.openmptcprouter.com buster/main amd64 DEP-11 Metadata
Ign:10 https://repo.openmptcprouter.com buster/main all DEP-11 Metadata
Ign:11 https://repo.openmptcprouter.com buster/main DEP-11 48x48 Icons
Ign:13 https://repo.openmptcprouter.com buster/main DEP-11 64x64 Icons
Ign:14 https://repo.openmptcprouter.com buster/main DEP-11 64x64@2 Icons
Ign:15 https://repo.openmptcprouter.com buster/main all c-n-f Metadata
Ign:16 https://repo.openmptcprouter.com buster/main amd64 c-n-f Metadata
Ign:17 https://repo.openmptcprouter.com buster/dists amd64 Packages
Ign:19 https://repo.openmptcprouter.com buster/dists all Packages
Reading package lists... Done
E: Failed to fetch https://repo.openmptcprouter.com/dists/buster/main/binary-amd64/Packages  Certificate verification failed: The certificate is NOT trusted. The certificate chain uses expired certificate.  Could not handshake: Error in the certificate verification. [IP: 68.70.205.3 443]
E: Some index files failed to download. They have been ignored, or old ones used instead.

tried adding 'dists' and "trusted=yes" to /etc/apt/sources.list.d/openmptcprouter.list, but no effect.

root@OMR-VPS:~# uname -a
Linux OMR-VPS 5.13.0-44-generic #49~20.04.1-Ubuntu SMP Wed May 18 18:44:28 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux

root@OMR-VPS:/etc/apt/sources.list.d# cat openmptcprouter.list
deb [arch=amd64 trusted=yes] https://repo.openmptcprouter.com buster main dists

not very knowledgable on Unix install stuff, so I'm kinda lost here...

Ysurac, 2022-05-29:
Check date/time on your VPS, if it's ok update OS with ```apt-get update && apt-get dist-upgrade```
(and you should use Debian)

packlet, 2022-05-29:
thanx, but doesn't seem to help.    is Ubuntu no longer supported/working?

root@OMR-VPS:~# date
Sun 29 May 2022 01:50:10 PM PDT

root@OMR-VPS:~# apt-get update && apt-get dist-upgrade
Get:1 http://security.ubuntu.com/ubuntu focal-security InRelease [114 kB]
Hit:2 http://us.archive.ubuntu.com/ubuntu focal InRelease                                                              
...
Ign:37 https://repo.openmptcprouter.com buster/dists all c-n-f Metadata
Ign:38 https://repo.openmptcprouter.com buster/dists amd64 c-n-f Metadata
Err:17 https://repo.openmptcprouter.com buster/main amd64 Packages
  Certificate verification failed: The certificate is NOT trusted. The certificate chain uses expired certificate.  Could not handshake: Error in the certificate verification. [IP: 68.70.205.1 443]
Ign:18 https://repo.openmptcprouter.com buster/main all Packages
Ign:19 https://repo.openmptcprouter.com buster/main Translation-en
Ign:20 https://repo.openmptcprouter.com buster/main Translation-en_US
Ign:21 https://repo.openmptcprouter.com buster/main amd64 DEP-11 Metadata
Ign:22 https://repo.openmptcprouter.com buster/main all DEP-11 Metadata
Ign:23 https://repo.openmptcprouter.com buster/main DEP-11 48x48 Icons
Ign:24 https://repo.openmptcprouter.com buster/main DEP-11 64x64 Icons
Ign:25 https://repo.openmptcprouter.com buster/main DEP-11 64x64@2 Icons
Ign:26 https://repo.openmptcprouter.com buster/main all c-n-f Metadata
Ign:27 https://repo.openmptcprouter.com buster/main amd64 c-n-f Metadata
Fetched 1,228 kB in 6s (208 kB/s)
Reading package lists... Done
E: Failed to fetch https://repo.openmptcprouter.com/dists/buster/main/binary-amd64/Packages  Certificate verification failed: The certificate is NOT trusted. The certificate chain uses expired certificate.  Could not handshake: Error in the certificate verification. [IP: 68.70.205.1 443]
E: Some index files failed to download. They have been ignored, or old ones used instead.
root@OMR-VPS:~#

Ysurac, 2022-05-29:
Ubuntu is supported, but I think you have a not updated release with old CA certificates.
Run the command before running the script, so reinstall or delete the repo from list.

packlet, 2022-05-29:
thanx!  re-installed fresh 20.04.1, and updated before running the VPS install script.  seems to have installed now.   
(I had assumed the 'install updates' option offered by install wizard would do that, but i guess not...)

closed 2022-09-03 by github-actions[bot]