Ysurac/openmptcprouter #2370: OMR-Tracker recieves no response from Non-Master link gateway

pbix, 2022-05-30:
## Expected Behavior

The OMR-Tracker should be able to register ping responses from all link gateways.

## Current Behavior

The non-master link of a two link system often shows no response.  
At the same time manual pings from the command line have no issues receiving responses as shown below.
The display says "No response after 1 second" but I have no "1 second" timeouts.  All my timeouts are set higher.

## Possible Solution

No idea.

## Steps to Reproduce the Problem

  1.
  2.
  3.

## Context (Environment)

It seems impossible to me for OpenMPTCPRouter to work correctly if it does not receive reliable ping responses.
I also see the following strange message with dmesg.  I do not know what it means.
"__mptcp_init4_subsockets: token 0xbf4fcf51 bind() to 192.168.100.100 index 52 failed, error -99"

## Specifications

  - OpenMPTCProuter version: v58.4 
  - OpenMPTCProuter VPS version: (how to know?) 
  - OpenMPTCProuter VPS provider: linode
  - OpenMPTCProuter platform: RPi3 
  - Country: USA 

[image]
[image]
[image]

Ysurac, 2022-05-30:
Status page is not related at all with omr-tracker. 
Warning you can see on status page are from tests done for status page only. There is in System->OpenMPTCProuter, "advanced settings" tab some timeout that can be defined but it's only for status page display.
For your message, what is 192.168.100.100 ip ? what is the result of ```ip a```?

pbix, 2022-05-30:
I can see that my master link is being switched off in the logs so I think it is more than just a status display issue.
Also, I found the "advanced settings" tab.  There is no "1 second there" so why does it show this "1 second" on the status screen?

[image]

Ysurac, 2022-05-30:
On the log, the only up/down of the port is due to a detected cable disconnection (and using bridge is not recommended).
There is "VPS checks timeout" in advanced settings tab. But as I already said it's only a status page test, nothing is done if the test fail only displayed message.
I don't see any 192.168.100.100 ? an old IP ?

pbix, 2022-05-30:
I have these messages in the log.  They are due to OMR-Tracker I think?

Mon May 30 14:57:05 2022 user.notice post-tracking-post-tracking: wan1 (eth0.3) switched up
Mon May 30 14:57:59 2022 user.notice post-tracking-post-tracking: wan1 (eth0.3) switched off because check error and ping from 198.18.61.157 error (1xx.xxx.xxx.174,1xx.xxx.xxx.174)
Mon May 30 14:57:59 2022 user.notice post-tracking-post-tracking: Delete default route via 198.18.61.1 dev eth0.3
Mon May 30 14:59:08 2022 daemon.notice netifd: wan2 (28119): udhcpc: sending renew to 100.64.0.1
Mon May 30 14:59:08 2022 daemon.notice netifd: wan2 (28119): udhcpc: lease of 100.94.71.129 obtained, lease time 300
Mon May 30 14:59:50 2022 user.notice post-tracking-post-tracking: wan1 (eth0.3) switched up
Mon May 30 15:00:30 2022 user.notice post-tracking-post-tracking: wan1 (eth0.3) switched off because check error and ping from 198.18.61.157 error (1xx.xxx.xxx.174,1xx.xxx.xxx.174)
Mon May 30 15:00:30 2022 user.notice post-tracking-post-tracking: Delete default route via 198.18.61.1 dev eth0.3
Mon May 30 15:01:38 2022 daemon.notice netifd: wan2 (28119): udhcpc: sending renew to 100.64.0.1
Mon May 30 15:01:38 2022 daemon.notice netifd: wan2 (28119): udhcpc: lease of 100.94.71.129 obtained, lease time 300
Mon May 30 15:02:36 2022 user.notice post-tracking-post-tracking: wan1 (eth0.3) switched up
Mon May 30 15:03:18 2022 user.notice post-tracking-post-tracking: wan1 (eth0.3) switched off because check error and ping from 198.18.61.157 error (1xx.xxx.xxx.174)
Mon May 30 15:03:18 2022 user.notice post-tracking-post-tracking: Delete default route via 198.18.61.1 dev eth0.3
Mon May 30 15:04:08 2022 daemon.notice netifd: wan2 (28119): udhcpc: sending renew to 100.64.0.1

Ysurac, 2022-05-30:
Yes, these message are due to omr-tracker.
It seems to have problem to ping VPS but pinging VPS shouldn't be the only test... You modified omr-tracker config ?

pbix, 2022-05-30:
And I do not know what 192.168.100.100 is.  I have never had such a net in my system.  But it may be that the Starlink system starts with this IP on this subnet on eth0.2 before switching to the IP it gets via DHCP from the network.

I have changed some of the settings for omr-tracker config on the webpage.  That is all.
[image]
[image]

pbix, 2022-05-30:
I have noticed that is can take about 3 seconds for the first ping response on eth0.3.  Not sure why.  After that it pings a 1 second intervals.  So I am wondering if it fails in this initial interval.

Ysurac, 2022-05-30:
You modified Hosts variable. Hosts are used for WAN connection test, in your case only test to one IP can be done, it's a problem.

pbix, 2022-05-30:
Initially I had three popular DNS IP addresses there.  Now it is showing my VPS only.  It seems to make no different with way.  I will switch it back though if you recommend.

So there is no way to change the "1 second" on the status screen?

Ysurac, 2022-05-30:
There is "Disable server ping" in advanced settings tab, but it's more for server with ICMP blocked.

pbix, 2022-05-30:
In OpenWRT we had luci-app-statistics and collectd.  They help to evaluate the stability of links in read time.
Can those be installed under OMR?

Ysurac, 2022-05-30:
They can be installed via System->software

pbix, 2022-05-30:
The following errors are displayed if you try and install luci-app-statistics

    Required dependency package luci-lib-nixio is not available in any repository.
    Required dependency package luci-lib-jsonc is not available in any repository.
    Required dependency package libgcc1 is not available in any repository.

Ysurac, 2022-05-31:
You need to follow https://github.com/Ysurac/openmptcprouter/issues/2099 to update package list.

pbix, 2022-06-01:
> You need to follow #2099 to update package list.

Thanks for the tip, that fixed it.