Ysurac/openmptcprouter #2375: No internet after rebooting OpenMCTPRouter

voarsh2, 2022-06-01:
## Expected Behavior
Rebooting should have internet as usual

## Current Behavior
When rebooting, I find the following issue:
[image]

The only resolution is to remove all the current WAN configs in openmptcprouter/wizard and re-add them and all is normal again...

## Specifications

  - OpenMPTCProuter version: [v0.58.5 r0+16336-b36068d35d](https://github.com/ysurac/openmptcprouter)
  - OpenMPTCProuter VPS version: Version 0.1026 5.4.100-mptcp
  - OpenMPTCProuter VPS provider: Contabo
  - OpenMPTCProuter platform: x86_64)

Ysurac, 2022-06-02:
Do you use MacVlan ? What are the type of connections used ? ethernet or usb ?
What do you have in Network->Interfaces ?
The host platform is x86_64, it's a VM ? something else ?

voarsh2, 2022-06-02:
> Do you use MacVlan ? What are the type of connections used ? ethernet or usb ? What do you have in Network->Interfaces ? The host platform is x86_64, it's a VM ? something else ?

It's a VM.

> What do you have in Network->Interfaces ?

I would have to reproduce it again for a screenshot, which I am not keen on since I would have no internet for a bit.... but I did happen to look at the interfaces, they were all there. I even tried restarting them... which didn't work, so I had to remove them from setup wizard and re-add them.

> Do you use MacVlan ? What are the type of connections used ? ethernet or usb ?

Ethernet, using "normal" mode, not MacVlan.

Ysurac, 2022-06-09:
If this happen again, I would need the result of ```uci show network```, ```ip a``` and ```ip r``` via SSH on the router.

closed 2022-09-12 by github-actions[bot]