Ysurac/openmptcprouter #2379: DOCS - Provide recommanded hardware

storm1er, 2022-06-05:
**Is your enhancement proposal related to a problem? Please describe.**

I'm using an RPi4B since almost 2 years now with OMR and 3 4G router (no wired WAN) and willing to upgrade my setup

<details>
RPi4B
2 USB to eth dongle
2 4G router, permanently connected
1 USB to Android (tethering mode) connected when needed
1 dumb eth switch on lan port
1 dumb wifi hotspot on switch
</details>

**Describe the solution you'd like**

Documentation enhancement giving the basic but perfect setup, including multiple links to Amazon/LDLC/Other retailers for the given recommanded hardware

To me, basic is:
- 1 router (Wan & Lan manager)
- 1 modem (Wan provider)
- 1 switch (Lan provider)
- eventually 1 wifi hotspot (if not included in router)
- eventually cables and dongles (if needed)

**Describe alternatives you've considered**

Based on this wiki page, I could guess recommanded hardware for the router, but not for everything else

https://github.com/Ysurac/openmptcprouter/wiki/Router-install

**Additional context**

1. This part of the documentation would be for people willing to get a basic setup working for themself, including *so called noobs*. The goal is really to provide one and only one basic setup

2. It could became the project reference with - long term talking here - official paid support, providing you a more robust way to get some money so we can support your project even more be feeling a part of it :)

Ysurac, 2022-06-09:
Would be cool, but difficult to recommand hardware for me. Needed hardware depends of connection number, quality and speed.

In most case for a quite fast speed I would recommand a x86_64 with AES-NI based hardware for router with fast core (better to have a fast core than many cores), anything that is ethernet for modem, any working switch (ubiquity have reported issues), manageable if you want to use VLAN.

RPI4 are actually costly, so I think more widely available cheap x86_64 barebones would work better but I don't any to test and recommend.

storm1er, 2022-06-09:
> 1. Would be cool, but difficult to recommand hardware for me. Needed hardware depends of connection number, quality and speed.

Well, as I said, the idea would be to provide "basic" stuff for newbies who want a better connection.

I'll take the 3 installs I made as an example and I'm pretty sure most "newbies" users would use it like this: 
- Using their current wired-but-bad ADSL
- enhance it with 1 or 2 4G connections
- enjoy the system as it is

Also, this probably matches 75+% of your current user base anyway

> 2. In most case for a quite fast speed I would recommand a x86_64 with AES-NI based hardware for router with fast core

What are you using as a router for yourself, (what's the one having the most well-tested development you made for this project)?
Would a simple Intel NUC do the job? (current cheapest being `Intel NUC NUC7CJYHN2`)

> 3. Working switch, manageable if you want to use VLAN

Manageable is a non-negligible plus, does a basic 8 ports Netgear GS308E would do the job?