Ysurac/openmptcprouter #2381: Using Pi-Hole with OMR-Tracker

patsandy100, 2022-06-08:
## Expected Behavior
When the vpn/vps server is offline or unavailable I would like traffic to use a public dns such as 1.1.1.1 etc but while everything is working everything go through Pihole dns setup on the VPS server.

## Current Behavior
Currently unsure how to set the DNS/DHCP section of Openmptcprouter. I have the ip of the pihole interface as primary and 1.1.1.1 as secondary but i appear to be having majority of dns requests going through 1.1.1.1 over pihole. Unsure what this entry is for "127.0.0.1#5353" does or is for. Am I even setting the dns forwarders in the right place?

## Specifications

  - OpenMPTCProuter version: [v0.58.5 r0+16336-b36068d35d]
  - OpenMPTCProuter VPS version: Rebuilt today with script. Debian 10
  - OpenMPTCProuter VPS provider: Vultr
  - OpenMPTCProuter platform: x86_64

Ysurac, 2022-06-09:
If you install pi-hole via the script in https://github.com/Ysurac/openmptcprouter/wiki/Pi-hole it should be detected and then used when you validate the wizard.
127.0.0.1#5353 relay DNS queries to unbound that then use root DNS servers to resolve the request.
Pi-Hole internal IP (via the VPN) can be set manually instead of 127.0.0.1#5353 in this place.

patsandy100, 2022-06-09:
Thanks Ysurac my apologies for the silly question. Might I just say I love your software its making my Starlink not a complete waste of time. Keep up the great work.
PS: I may open another ticket shortly regarding another issue.

closed 2022-06-09 by patsandy100