Ysurac/openmptcprouter #2383: Aggregation Bandwidth

tuanbombt, 2022-06-09:
## Expected Behavior
Currently I am aggregating bandwidth from 4 pppoe accounts.

1 line = 110 Mbps
4 lines = 440 Mbps

But after I check the bandwidth, it is only about 283 Mbps (both up / down lines are the same).

VPS I tried to test with speedtest, the maximum speed can be received is 980 Mbps (both up / down).

The problem here is that the aggregate bandwidth cannot pool more than 300 Mbps.

Current used Shadowsocks chacha20, glorytun udp.

MPTCP: 

Multipath TCP scheduler:Blest, 
Congestion Control: cubic and Multipath TCP SYN retries:2

## Specifications

  - OpenMPTCProuter version: OpenMPTCProuter v0.59 beta 6
  - OpenMPTCProuter VPS version: 5.4.196-mptcp 0.1027-test
  - OpenMPTCProuter VPS provider: Lightsail
  - OpenMPTCProuter platform: x86_64

Ysurac, 2022-06-09:
You can try another TCP Congestion Control.
What is the latencies you get in status page ?
What is the load average during a speedtest on both router and VPS ? Use "omr-test-speed" via SSH on the router to test aggregated speed.

tuanbombt, 2022-06-09:
> You can try another TCP Congestion Control. What is the latencies you get in status page ? What is the load average during a speedtest on both router and VPS ? Use "omr-test-speed" via SSH on the router to test aggregated speed.

Latency around 20-25ms

omr-test-speed, result around 20mb/s

About the change TCP Congestion Control, change both vps and router or only router?

Ysurac, 2022-06-09:
When you change on router congestion is changed on the VPS

tuanbombt, 2022-06-09:
Same, i think problem because VPS.

tuanbombt, 2022-06-09:
https://github.com/shadowsocks/shadowsocks-rust/pull/345#issuecomment-748118974
https://github.com/honwen/openwrt-shadowsocks-rust
https://github.com/honwen/luci-app-shadowsocks-rust

@Ysurac do you have you thought about changing shadowsocks-libev to shadowsocks-rust.

I looked at the benchmark between the two software, shadowsocks-rust performance is better.

tuanbombt, 2022-06-09:
@Ysurac 

After I tried shadowsocks-rust, Speed download/upload and ping very good

Download test file from DigitalOcean Sing = (50mb/s)

Test with omr-test-speed (around 40mb/s).

closed 2022-06-11 by tuanbombt