Ysurac/openmptcprouter #2384: Binding wireguard to specific WAN interface

SriramScorp, 2022-06-09:
Hey @Ysurac!

With regards to #1252 and #1314, how does wireguard work now? Did you figure out a way to make wireguard bind to an interface, since wireguard doesn't have such an option in-built, unlike openvpn?

Ysurac, 2022-06-10:
I use fwmark setting that mark all packets going out of wireguard, then with ip rule I attribute a routing table to the fwmark and I'm able to do what I want.

SriramScorp, 2022-06-10:
When I have one WAN interface (eth1) and a corresponding wireguard interface (wgeth1), the actual traffic to wireguard port (65311/udp) doesn't go through wgeth1, but through eth1 (confirmed via tcpdump). When there are 2 WAN interfaces (eth1, eth2) which filter MPTCP, and I need to use MPTCPoverVPN via wireguard for both, how do you run two instances of wireguard where the traffic to port 65311 flows through eth1 & eth2 respectively? Since the destination IP and port are the same, how do you distinguish b/w them to mark outgoing packets? Could you provide a sample snippet of the required configuration?

Wyk72, 2022-08-16:
You have to use VXLAN encapsulation into a wireguard tunnel for this to work.

VXLAN interfaces behave like standard eth interfaces (more or less...) so they can be bonded/bridged to your heart's content.

So you create the VXLAN endpoints into the wireguard interface you prefer,  building a "eth cable" into the tunnel of your choice, then you treat them as normal eth interfaces. No need for binding, as the tunnels tou create are specific to that wireguard link.

closed 2022-11-20 by github-actions[bot]

Wyk72, 2025-02-11:
As a clean "workaround", I use a namespace, send the ethernet link/interface to that namespace, then create  the wireguard link in that namespace. 

After that, I move the wireguard link/interface to he default "init" namespace. 

This way  Wireguard will always "talk" to that specific network interface.

This is described into the wireguard docs, here: https://www.wireguard.com/netns/ at section "The New Namespace Solution" and that's where I took inspiration from.

It works very well and it was a rather interesting exercise. Opens up a lot of possibilities.