Ysurac/openmptcprouter #2385: Two DHCP servers, tried blocking third party with IPTables

voarsh2, 2022-06-09:
## Expected Behavior

I added the following custom IPTable rules in the GUI:

```
iptables -A INPUT -p tcp --src 192.168.1.1 -m tcp -m multiport ! --dports 80,443 -j DROP
iptables --append INPUT --protocol udp --src 192.168.1.1 --sport 67 --jump DROP
iptables --append INPUT --protocol tcp--src 192.168.1.1 --sport 53 --jump DROP
iptables --append INPUT --protocol udp --src 192.168.1.1 --sport 68 --jump DROP
iptables --append FORWARD --protocol udp --src 192.168.1.1 --sport 68 --jump DROP
iptables --append FORWARD --protocol udp --src 192.168.1.1 --sport 67 --jump DROP
```
I've called it WAN 3, it has DHCP on, I am unable to turn it off as it is serving other users.
I thought this was sufficient to block all DHCP traffic from WAN 3. After a while OpenMPTCProuter would display not connected (but all non DHCP VM's wouldn't be able to access sites). 
My VM's inside Proxmox (OpenMPTCProuter is also in a VM) would start to show the WAN 3 in the Windows networking status (communityfibre.co.uk dns networking name).

I then became unsure if my traffic was even going through OpenMPTCProuter and my other two WAN networks. In my testing a new interface LAN using eth0 wouldn't

In Proxmox itself I set the following ebtables:
```
ebtables -I FORWARD 1 -p IPv4 --ip-src ! 192.168.100.1 --ip-proto udp --ip-sport 67 -j DROP
ebtables -I FORWARD 1 -p IPv4 --ip-src ! 192.168.100.1 --ip-proto udp --ip-sport 68 -j DROP

```

At one point OpenMPTCProuter rebooted and I am sure WAN 3 took over DHCP stuff for the few seconds it had restarted.
My networking switch has no firewall, so I can't firewall it at the hardware level

## Specifications

  - OpenMPTCProuter version: v0.58.5 r0+16336-b36068d35d](https://github.com/ysurac/openmptcprouter)
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform:

Ysurac, 2022-06-12:
You should use Network->Firewall interface, but I don't really understand what you want to do and why ;)

voarsh2, 2022-08-12:
> You should use Network->Firewall interface, but I don't really understand what you want to do and why ;)
I used custom rules, which did not help blocking DHCP.  Do you mean to try: Network -> Firewall -> Traffic Rules?

I had a third router that I don't want to be disabling DHCP on, as it is not mine to mess with settings - but I am able to connect to it.
Was hoping to utilise it and block DHCP.

closed 2022-11-16 by github-actions[bot]