Ysurac/openmptcprouter #2386: No internet connection when master WAN failed

Kalimeiro, 2022-06-10:
## Expected Behavior
When master WAN failed, we keep TCP/UDP/... connections, we can continue to go on website, we can resolve dns name, the VPN (Glorytun, mlvpn, ...) work without any drop.

Lorsque la connexion WAN maître tombe (désynchro VDSL, changement d'IP public, ...) je dois conserver les connexions TCP/UDP/..., actives et je dois pouvoir continuer de naviguer sur internet, résoudre des noms dns, accéder au VPN (Glorytun, mlvpn, ...) sans constater de déconnexion.

I'm not supposed to notice my connections going down this way, other than a drop in throughput, when connections is back i need to notice with a up in throughput.

Je ne suis pas censé remarquer de cette manière que ma connexion WAN maître ne fonctionne plus, hors une baisse de débit, lorsque la connexion WAN maître revient je dois constater une augmentation du débit.

## Current Behavior
When master WAN failed, i don't have internet for around 1 minute, i see glorytun and shadowsocks restart, i lost all TCP/UDP/... active connections.

Lorsque la connexion WAN maître tombe, je n'ai plus du tout internet pendant ~1 minute (le temps que la box se resynchro), je vois que glorytun et shadowsocks redémarre, je perds toutes connexions TCP/UDP/... actives.

## Steps to Reproduce the Problem

  1. Restart master WAN BOX or remove the wired ethernet of the master WAN.
  2. you'l see master WAN failed in OMR, but not WAN2/WAN3/...
  3. 30 seconds / 1 minute after, no internet and all active connections drop, include glorytun.

  1. Redémarrer la box WAN maître, ou enlever le câble réseau.
  2. On voit que la connexion WAN maître ne fonctionne plus dans OMR, mais pas WAN2/WAN3/...
  3. 30 secondes / 1 minute plus tard, il n'y a toujours pas accès à internet et toutes les connexions sont terminés, incluant le VPN Glorytun.

## Specifications

  - OpenMPTCProuter version: 0.59 beta6 (latest snapshot)
  - OpenMPTCProuter VPS version: 0.1027 (latest snapshot)
  - OpenMPTCProuter VPS provider: OVH
  - OpenMPTCProuter platform: RPI4/X86_64/R4S
  - Country: FR

Ysurac, 2022-06-10:
I would need, before wan failed, the result of ```ip a```, ```ip r``` and ```multipath -f```
And after wan failed, same commands + status->system log
What are you using ? shadowsocks + glorytun tcp as default ?

Kalimeiro, 2022-06-10:
I just update to 0.59 beta 7 and this problem is solved, perhaps the beta 6 i use have a problem and has been solved.
But, i have a new issue, so i will close this issue and open a new.
Thanks for all.

closed 2022-06-10 by Kalimeiro