Ysurac/openmptcprouter #2389: multiple sames rules in shorewall, he finished to crash and VPS not responding on the net.

Kalimeiro, 2022-06-12:
## Expected Behavior
When a port forward rules is create on OMR with filter by source IP, it will be create only one times on rules files on shorewall.

## Current Behavior
When a port forward rules is create on OMR with filter by source IP, the same rules is create at infinite, causing high CPU usage on VPS and on OMR, finally shorewall finish to crash and VPS do not respond, then on OMR all WAN appear down (see #2388)

## Steps to Reproduce the Problem

  1. create rules with filter by IP
  2. wait few hours or few days

## Specifications

  - OpenMPTCProuter version: 0.59 beta 7
  - OpenMPTCProuter VPS version: 0.1027-test
  - OpenMPTCProuter VPS provider: OVH
  - OpenMPTCProuter platform: RPI4
  - Country: FR

Ysurac, 2022-06-12:
I need the result of ```uci show firewall``` via SSH on the router, and the content of ```/etc/shorewall/rules``` on the VPS.

Kalimeiro, 2022-06-12:
i send it by mail at you to have confidentiality.

Ysurac, 2022-06-16:
I made some changes on VPS script that should fix this issue.

Kalimeiro, 2022-06-16:
it's partially solved i don't see multiple same rules, but when you play with enable/disable rules, it's ok when you enable/disable without filter by ip source, but when you have specific rules with ip source defined, it's not removed in shorewall.

Kalimeiro, 2022-06-16:
i have test more and i confirm for now if you removed or disable a rules filtered by ip source, it be keeped in shorewall rules.

Kalimeiro, 2022-06-17:
With the latest update of this day, the problem isn't solved.

Kalimeiro, 2022-06-18:
the problem is definitively solved with the latest typo fix. Thanks you.

closed 2022-06-18 by Kalimeiro