Ysurac/openmptcprouter #2390: Port forward not working

Bestle, 2022-06-12:
I am trying to forward port 9010 to a device on my network that is running a web server.  "tcpdump -i tun0 port 9010" on the router shows that traffic is arriving on that port, but the web server is not accessible from outside the LAN. I am not sure how to debug where the packets are actually ending up.

[image]

[image]

[image]

I will put a 50 GBP bounty on helping me get this resolved.

  - OpenMPTCProuter version: openmptcprouter v0.58.5 r0+16336-b36068d35d / LuCI Master git-21.077.59320-d1bf56d
  - OpenMPTCProuter VPS provider: linode
  - OpenMPTCProuter platform: x86_64
  - Country: UK

fxousselin, 2022-06-12:
Have you tried :
From VPN to lan ?

Ysurac, 2022-06-12:
Did you check the result of tcpdump on destination server ?

Bestle, 2022-06-12:
Some traffic does appear to arrive at the destination when looking with tcpdump and if it's relevant this is for a plex server. I have tried hosting an nginx on another local server and that is also unreachable.

`17:01:17.289862 IP NAS.lan.9010 > OpenMPTCProuter.lan.56845: Flags [S.], seq 633645868, ack 3027621759, win 64240, options [mss 1460,nop,nop,sackOK,nop,wscale 7], length 0
17:01:17.545824 IP NAS.lan.9010 > OpenMPTCProuter.lan.56846: Flags [S.], seq 2872921818, ack 178632665, win 64240, options [mss 1460,nop,nop,sackOK,nop,wscale 7], length 0
17:01:18.057840 IP NAS.lan.9010 > OpenMPTCProuter.lan.60897: Flags [S.], seq 3790626272, ack 2208253704, win 64240, options [mss 1460,nop,nop,sackOK,nop,wscale 7], length 0
17:01:18.313821 IP NAS.lan.9010 > OpenMPTCProuter.lan.60899: Flags [S.], seq 350642794, ack 2806525695, win 64240, options [mss 1460,nop,nop,sackOK,nop,wscale 7], length 0
17:01:25.663486 IP NAS.lan.9010 > TOM-PC.lan.53350: Flags [P.], seq 665588:665615, ack 46988, win 501, length 27
17:01:25.663581 IP TOM-PC.lan.53350 > NAS.lan.9010: Flags [R.], seq 46988, ack 665615, win 0, length 0
17:01:33.421729 IP NAS.lan.9010 > OpenMPTCProuter.lan.56845: Flags [S.], seq 633645868, ack 3027621759, win 64240, options [mss 1460,nop,nop,sackOK,nop,wscale 7], length 0
17:01:33.421735 IP NAS.lan.9010 > OpenMPTCProuter.lan.56844: Flags [S.], seq 4132146775, ack 2755242740, win 64240, options [mss 1460,nop,nop,sackOK,nop,wscale 7], length 0
17:01:33.673816 IP NAS.lan.9010 > OpenMPTCProuter.lan.56846: Flags [S.], seq 2872921818, ack 178632665, win 64240, options [mss 1460,nop,nop,sackOK,nop,wscale 7], length 0
17:01:51.415942 IP OpenMPTCProuter.lan.44740 > NAS.lan.9010: Flags [S], seq 623145713, win 42300, options [mss 1410,sackOK,TS val 1591353558 ecr 0,nop,wscale 12], length 0
17:01:51.416086 IP NAS.lan.9010 > OpenMPTCProuter.lan.44740: Flags [S.], seq 28771813, ack 623145714, win 65160, options [mss 1460,sackOK,TS val 679225037 ecr 1591353558,nop,wscale 7], length 0
17:01:52.425774 IP NAS.lan.9010 > OpenMPTCProuter.lan.44740: Flags [S.], seq 28771813, ack 623145714, win 65160, options [mss 1460,sackOK,TS val 679226047 ecr 1591353558,nop,wscale 7], length 0
17:01:52.450606 IP OpenMPTCProuter.lan.44740 > NAS.lan.9010: Flags [S], seq 623145713, win 42300, options [mss 1410,sackOK,TS val 1591354592 ecr 0,nop,wscale 12], length 0
17:01:52.450700 IP NAS.lan.9010 > OpenMPTCProuter.lan.44740: Flags [S.], seq 28771813, ack 623145714, win 65160, options [mss 1460,sackOK,TS val 679226072 ecr 1591353558,nop,wscale 7], length 0
17:01:54.462187 IP OpenMPTCProuter.lan.44740 > NAS.lan.9010: Flags [S], seq 623145713, win 42300, options [mss 1410,sackOK,TS val 1591356604 ecr 0,nop,wscale 12], length 0
17:01:54.462295 IP NAS.lan.9010 > OpenMPTCProuter.lan.44740: Flags [S.], seq 28771813, ack 623145714, win 65160, options [mss 1460,sackOK,TS val 679228083 ecr 1591353558,nop,wscale 7], length 0
17:01:56.489830 IP NAS.lan.9010 > OpenMPTCProuter.lan.44740: Flags [S.], seq 28771813, ack 623145714, win 65160, options [mss 1460,sackOK,TS val 679230111 ecr 1591353558,nop,wscale 7], length 0`

Bestle, 2022-06-14:
In the end I "fixed" this with a fresh install on the router. Couldn't figure it out.

closed 2022-06-14 by Bestle