Ysurac/openmptcprouter #2393: On VPS Restart, Pi-hole is unreachable and certain ports are blocked

darkmaniac7, 2022-06-17:
## Expected Behavior
After VPS Restart All services and ports are still reachable

## Current Behavior
After VPS Restart Pi-hole is unreachable and certain ports are blocked, ICMP is blocked on WAN for example (e.g. ping-t 8.8.8.8)
Also issue with VPN (anyconnect) after restarting VPS
Web and HTTPS, SSH are all still working.

## Specifications

  - OpenMPTCProuter: v0.59beta7-5.4
  -  OpenMPTCProuter VPS: 0.1026
  - OpenMPTCProuter VPS provider: Kamatera
  - OpenMPTCProuter platform: x86_64

So far the only fix I've found for the problem is when needing to update/upgrade or perform a restart creating a new VPS and reinstalling PiHole & OMR VPS.

Ysurac, 2022-06-29:
Pi-Hole doesn't make any change on firewall.
For the VPN, where did you set it ? on client, router or VPS ?
By default, OpenMPTCProuter use a proxy (shadowsocks) for TCP and a VPN (glorytun tcp) for anything else (UDP, ICMP,...)

darkmaniac7, 2022-06-30:
> on client,

Thanks for the Reply,

For VPN it is on the Client, the VPN is for connecting to Work.

I changed from Shadowsocks to V2Ray if that makes any difference. I thought maybe the PiHole issue may be related to this?

https://github.com/Ysurac/openmptcprouter/issues/1179

I haven't yet tried if the same issue happens (Anyconnect VPN & ICMP Ports not responding) without Pi-Hole. I'll try that this weekend.

darkmaniac7, 2022-07-09:
After upgrading to the latest version 0.59 Beta 8 I think it may have to do with upgrading and the firewall.

Upon the previous upgrade from 0.59 Beta 6 to 7 I also updated the VPS and restarted it. I figured the restart is what caused the issue with Pinging, Port Forwarding, VPN, PiHole, etc.

This time upon upgrade the same thing happened, but I didn't upgrade/update/restart the VPS. My fix was to delete all of the firewall rules, and add them back one by one saving on each change. Afterwards it worked. Pinging, Web Server Reachable, Plex, Port Forwarding, VPN, PiHole, etc.

To test it again, I may reboot the VPS tonight and see if the same issue happens and if reloading the Firewall rules fixes it.

closed 2022-07-20 by darkmaniac7