Ysurac/openmptcprouter #2398: add avahi-daemon

dr1pvfx, 2022-06-19:
**Is your feature request related to a problem? Please describe.**

I want to be able to resolve devices inside my IOT VLAN from my trusted LAN.

**Describe the solution you'd like**

add avahi-deamon to openmptcprouter

**Describe alternatives you've considered**
Manually install avahi-daemon but `opkg install avahi-daemon ` doesnt work and I am scared to break everything if I dig further. 

**Additional context**

https://blog.christophersmart.com/2020/03/30/resolving-mdns-across-vlans-with-avahi-on-openwrt/

winkmichael, 2022-06-22:
What you actually want is to enable proxy arp

echo 1 > /proc/sys/net/ipv4/conf/eth0/proxy_arp

replace eth0 with the interface of your main network and your trusted vlan

dr1pvfx, 2022-06-22:
thats very interesting, is there anything more like do I need to ipup/down the interfaces for that to work? if I run arp -a on a devide in eth0 i dont get any arps from the iOT vlan.

closed 2022-09-25 by github-actions[bot]