Ysurac/openmptcprouter #2401: Wifi calling is not supported

DeLoWaN, 2022-06-20:
## Expected Behavior
Wifi calling is working when using OMR supported.

## Current Behavior
In my house I have very poor mobile network reception, I then use wifi calling to handle the calls on my phone (Huawei P30, Android 12.0.0.134). It works well when I set static IP to both of my WAN, but whenever I use OMR as a router, the wifi calling feature refuses to work.

Interestingly, I do not have the issue with an iPhone.

## Possible Solution

## Steps to Reproduce the Problem

  1. Activate the wifi calling feature on an Android Phone (the carrier should authorize it)

## Context (Environment)

## Specifications

  - OpenMPTCProuter version: v0.58.5 r0+16336-b36068d35d
  - OpenMPTCProuter VPS version: 5.4.100-mptcp 0.1026
  - OpenMPTCProuter VPS provider: Scaleway
  - OpenMPTCProuter platform: x86_64
  - Country: France

Ysurac, 2022-06-22:
You mean VoWiFi ? I don't have a compatible phone to test that for now...
Strange that it work with iPhone, are you sure you output via OMR in this case ?

DeLoWaN, 2022-06-23:
> You mean VoWiFi ? I don't have a compatible phone to test that for now... Strange that it work with iPhone, are you sure you output via OMR in this case ?

Yes, I meant VoWIFI. Yes, I'm sure OMR is in use, I use OMR subnetwork IP (192.168.100.0/24) to test. If I set the IP on my phone of my subnetwork ADSL (192.168.10.0/24) or subnetwork 4G (192.168.5.0/24), it works. And yes, that is odd that for iPhone it works, and I only have one Android phone VoWIFI compatible to test at my home. Maybe when friends or relatives comes over, I will have some more devices available for testing :).

Actually it does not bother me that much since I just set the IP of my ADSL subnetwork on my phone, I do not need aggregation on it, but I though it was a good idea to report it if anyone already encountered the issue.

DeLoWaN, 2022-06-23:
Ah, I found something interesting. I read [this comment](https://github.com/Ysurac/openmptcprouter/issues/2109#issuecomment-941158721) which tells that with V2ray, it was not working.

I confirm it comes from V2ray, since I enabled it following the [optimisation](https://github.com/Ysurac/openmptcprouter/wiki/Optimization) wiki. Once disabled and back to Shadowsocks, the Wifi works now on my Android phone.

Maybe this info should go on a doc somewhere.

Jimmy062006, 2022-06-27:
It works fine over OMR for my Android phone and WiFi calling

closed 2022-10-01 by github-actions[bot]