Ysurac/openmptcprouter #2402: When update OMR, my virtual interface for VPN SITE-TO-SITE lost her firewall zone.

Kalimeiro, 2022-06-20:
## Expected Behavior
New interfaces created and assigned to a firewall zone need to be keeped where upgrade OMR.

[image]

## Current Behavior
My new interface "VPN" lost her firewall zone configuration when upgrade OMR.

[image]

## Context (Environment)
The new interface and the firewall zone is for a VPN SITE-TO-SITE i have made with integrated OpenVPN, work great.

## Specifications

  - OpenMPTCProuter version: 0.59 beta 7
  - OpenMPTCProuter VPS version: 0.1027-test
  - OpenMPTCProuter VPS provider: OVH
  - OpenMPTCProuter platform: ALL RPI
  - Country: FR

Ysurac, 2022-06-20:
What is the initial firewall zone ?

Kalimeiro, 2022-06-20:
i use the same VPN zone of OMR for omrvpn and omr6in4

Kalimeiro, 2022-07-19:
bug still exist with 0.59rc1

Ysurac, 2022-07-19:
You should use another zone, VPN zone is really for internal usage, maybe I should rename it to InternalVPN or something like that.

Kalimeiro, 2022-09-13:
ok but if i create/rename the zone name, i need to recreate most of NAT rules to have a good vpn site-to-site working, this is the reason i have set to the zone VPN.

Kalimeiro, 2022-09-13:
i close the issue, i have spend some times to create correctly a new zone and it's ok after upgrade to the final release 0.59.1

closed 2022-09-13 by Kalimeiro