Ysurac/openmptcprouter #2407: iPhone not working in Raspberry Pi 3b+

nscpro, 2022-06-26:
Hi.

When I connect the iPhone to the Raspberry Pi USB port. The terminal indicates that the iPhone is connected and the iPhone has also pressed the Trust button. If I want to add an iPhone, I don't see USB0 in the selection. Also, my phone doesn't show that Personal Hotspot is in use, although it's always on.

What should I do to get the iPhone to work?

I'm using latest openmptcprouter 64-bit kernel in Raspberry Pi 3b+

fsufyan, 2022-06-26:
I have been using iPhone usb tethering with raspberry pi 3B+ normally for weeks now.
The interface name will show as eth1 or something like, usb0 is for android phones.

Did you create the interface and set the mptcp option as (enabled) in advanced settings?

vinceducat, 2022-06-27:
@nscpro  ther are some post in the issue about that, take a search ...

closed 2022-06-29 by nscpro