Ysurac/openmptcprouter #2409: Building OMR for unsupported platform failed

rdmitry0911, 2022-07-01:
Hi guys, I'm trying to build kernel 5.15 based OMR for ath79 platform for Xiaomi AIoT Router AC2350. The building processs fails almost all the time while applying 5.10 backport patches like this one:
```
Applying /home/rdmitry0911/projects/openmptcprouter/ath79/5.15/source/target/linux/generic/backport-5.10/850-v5.17-0003-PCI-aardvark-Add-support-for-DEVCAP2-DEVCTL2-LNKCAP2.patch using plaintext: 
patching file drivers/pci/controller/pci-aardvark.c
Hunk #1 FAILED at 883.
Hunk #2 succeeded at 584 (offset -314 lines).
Hunk #3 succeeded at 605 (offset -314 lines).
1 out of 3 hunks FAILED -- saving rejects to file drivers/pci/controller/pci-aardvark.c.rej
Patch failed!  Please fix /home/rdmitry0911/projects/openmptcprouter/ath79/5.15/source/target/linux/generic/backport-5.10/850-v5.17-0003-PCI-aardvark-Add-support-for-DEVCAP2-DEVCTL2-LNKCAP2.patch!
```
The question is why do we need these backport patches? Is there any way to avoid this patching (may be with less OMR functionality)?

Ysurac, 2022-07-01:
I think it's compiling kernel 5.10. You may need to modify _target/linux/ath79/Makefile_ to set **KERNEL_PATCHVER:=5.15**

rdmitry0911, 2022-07-02:
> 

> I think it's compiling kernel 5.10. You may need to modify _target/linux/ath79/Makefile_ to set **KERNEL_PATCHVER:=5.15**

Thank you for the tip. All patching problems disappeared, but build process still fails:
```
/home/rdmitry0911/projects/openmptcprouter/ath79/5.15/source/build_dir/target-mips_24kc_musl/linux-ath79_generic/ndpi-netfilter2-4-59f295ba86110f2642f76eb967b041c4108fedd5/ndpi-netfilter/src/../../src/lib/protocols/fiesta.c: In function 'ndpi_search_fiesta':
/home/rdmitry0911/projects/openmptcprouter/ath79/5.15/source/build_dir/target-mips_24kc_musl/linux-ath79_generic/ndpi-netfilter2-4-59f295ba86110f2642f76eb967b041c4108fedd5/ndpi-netfilter/src/../../src/include/ndpi_define.h:352:23: error: implicit declaration of function 'bswap_16'; did you mean 'swab16'? [-Werror=implicit-function-declaration]
  352 | # define get_l16(X,O) bswap_16(get_u_int16_t(X,O))
      |                       ^~~~~~~~
/home/rdmitry0911/projects/openmptcprouter/ath79/5.15/source/build_dir/target-mips_24kc_musl/linux-ath79_generic/ndpi-netfilter2-4-59f295ba86110f2642f76eb967b041c4108fedd5/ndpi-netfilter/src/../../src/lib/protocols/fiesta.c:55:36: note: in expansion of macro 'get_l16'
   55 |                                 && get_l16(packet->payload, 1) == packet->payload_packet_len - 3))) {
      |                                    ^~~~~~~
cc1: some warnings being treated as errors
make[7]: *** [scripts/Makefile.build:277: /home/rdmitry0911/projects/openmptcprouter/ath79/5.15/source/build_dir/target-mips_24kc_musl/linux-ath79_generic/ndpi-netfilter2-4-59f295ba86110f2642f76eb967b041c4108fedd5/ndpi-netfilter/src/../../src/lib/protocols/fiesta.o] Error 1
make[6]: *** [Makefile:1868: /home/rdmitry0911/projects/openmptcprouter/ath79/5.15/source/build_dir/target-mips_24kc_musl/linux-ath79_generic/ndpi-netfilter2-4-59f295ba86110f2642f76eb967b041c4108fedd5/ndpi-netfilter/src] Error 2
make[6]: Leaving directory '/home/rdmitry0911/projects/openmptcprouter/ath79/5.15/source/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-5.15.36'
make[5]: *** [Makefile:43: modules] Error 2
make[5]: Leaving directory '/home/rdmitry0911/projects/openmptcprouter/ath79/5.15/source/build_dir/target-mips_24kc_musl/linux-ath79_generic/ndpi-netfilter2-4-59f295ba86110f2642f76eb967b041c4108fedd5/ndpi-netfilter/src'
make[4]: *** [Makefile:3: all] Error 2
make[4]: Leaving directory '/home/rdmitry0911/projects/openmptcprouter/ath79/5.15/source/build_dir/target-mips_24kc_musl/linux-ath79_generic/ndpi-netfilter2-4-59f295ba86110f2642f76eb967b041c4108fedd5/ndpi-netfilter'
make[3]: *** [Makefile:91: /home/rdmitry0911/projects/openmptcprouter/ath79/5.15/source/build_dir/target-mips_24kc_musl/linux-ath79_generic/ndpi-netfilter2-4-59f295ba86110f2642f76eb967b041c4108fedd5/.built] Error 2
make[3]: Leaving directory '/home/rdmitry0911/projects/openmptcprouter/feeds/openmptcprouter/ndpi-netfilter2'
time: package/feeds/openmptcprouter/ndpi-netfilter2/compile#71.96#7.66#78.97
    ERROR: package/feeds/openmptcprouter/ndpi-netfilter2 failed to build.
make[2]: *** [package/Makefile:116: package/feeds/openmptcprouter/ndpi-netfilter2/compile] Error 1
make[2]: Leaving directory '/home/rdmitry0911/projects/openmptcprouter/ath79/5.15/source'
make[1]: *** [package/Makefile:110: /home/rdmitry0911/projects/openmptcprouter/ath79/5.15/source/staging_dir/target-mips_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/rdmitry0911/projects/openmptcprouter/ath79/5.15/source'
make: *** [/home/rdmitry0911/projects/openmptcprouter/ath79/5.15/source/include/toplevel.mk:230: world] Error 2
```
 Any idea how to get rid of this error?
Thanks in advance.

Ysurac, 2022-07-02:
You can disable ndpi-netfilter2, it's used by omr-bypass. Or you can try to fix the issue.

rdmitry0911, 2022-07-02:
> You can disable ndpi-netfilter2, it's used by omr-bypass. Or you can try to fix the issue.

Ok, thank you again. Now, when I disabled ndpi-netfilter2, the build process passed. However, the size of the squashfs image is too big for the 16M router footprint. I faced with the same problem in the past, when router footprint was not enough to store all the required packages. I managed to make a root overlay to a network disk and load everything from there. To resolve lack of space problem now I have to make all the omr required modules as loadable {M}. Is there anything special I should keep inside the image, or I can put everything to a network drive?
```

[========================================================================================================================================================================================|] 9385/9385 100%

Exportable Squashfs 4.0 filesystem, xz compressed, data block size 262144
	compressed data, compressed metadata, compressed fragments,
	no xattrs, compressed ids
	duplicates are removed
Filesystem size 42228.19 Kbytes (41.24 Mbytes)
	23.83% of uncompressed filesystem size (177209.24 Kbytes)
Inode table size 61226 bytes (59.79 Kbytes)
	18.59% of uncompressed inode table size (329341 bytes)
Directory table size 84550 bytes (82.57 Kbytes)
	33.53% of uncompressed directory table size (252190 bytes)
Number of duplicate files found 1769
Number of inodes 10000
Number of files 8951
Number of fragments 235
Number of symbolic links  560
Number of device nodes 1
Number of fifo nodes 0
Number of socket nodes 0
Number of directories 488
Number of ids (unique uids + gids) 1
Number of uids 1
	root (0)
Number of gids 1
	root (0)
[ -f /home/rdmitry0911/projects/openmptcprouter/ath79/5.15/source/build_dir/target-mips_24kc_musl/linux-ath79_generic/xiaomi_aiot-ac2350-kernel.bin -a -f /home/rdmitry0911/projects/openmptcprouter/ath79/5.15/source/build_dir/target-mips_24kc_musl/linux-ath79_generic/root.squashfs ]
dd if=/home/rdmitry0911/projects/openmptcprouter/ath79/5.15/source/build_dir/target-mips_24kc_musl/linux-ath79_generic/xiaomi_aiot-ac2350-kernel.bin >> /home/rdmitry0911/projects/openmptcprouter/ath79/5.15/source/build_dir/target-mips_24kc_musl/linux-ath79_generic/tmp/openmptcprouter-v0.59beta7-v0.59beta7-c4157bfc-r0+19602-cb1dc49c18-ath79-generic-xiaomi_aiot-ac2350-squashfs-sysupgrade.bin
4889+1 records in
4889+1 records out
2503267 bytes (2.5 MB, 2.4 MiB) copied, 0.0131216 s, 191 MB/s
dd if=/home/rdmitry0911/projects/openmptcprouter/ath79/5.15/source/build_dir/target-mips_24kc_musl/linux-ath79_generic/tmp/openmptcprouter-v0.59beta7-v0.59beta7-c4157bfc-r0+19602-cb1dc49c18-ath79-generic-xiaomi_aiot-ac2350-squashfs-sysupgrade.bin of=/home/rdmitry0911/projects/openmptcprouter/ath79/5.15/source/build_dir/target-mips_24kc_musl/linux-ath79_generic/tmp/openmptcprouter-v0.59beta7-v0.59beta7-c4157bfc-r0+19602-cb1dc49c18-ath79-generic-xiaomi_aiot-ac2350-squashfs-sysupgrade.bin.new bs=64k conv=sync
38+1 records in
39+0 records out
2555904 bytes (2.6 MB, 2.4 MiB) copied, 0.00158781 s, 1.6 GB/s
mv /home/rdmitry0911/projects/openmptcprouter/ath79/5.15/source/build_dir/target-mips_24kc_musl/linux-ath79_generic/tmp/openmptcprouter-v0.59beta7-v0.59beta7-c4157bfc-r0+19602-cb1dc49c18-ath79-generic-xiaomi_aiot-ac2350-squashfs-sysupgrade.bin.new /home/rdmitry0911/projects/openmptcprouter/ath79/5.15/source/build_dir/target-mips_24kc_musl/linux-ath79_generic/tmp/openmptcprouter-v0.59beta7-v0.59beta7-c4157bfc-r0+19602-cb1dc49c18-ath79-generic-xiaomi_aiot-ac2350-squashfs-sysupgrade.bin
dd if=/home/rdmitry0911/projects/openmptcprouter/ath79/5.15/source/build_dir/target-mips_24kc_musl/linux-ath79_generic/root.squashfs >> /home/rdmitry0911/projects/openmptcprouter/ath79/5.15/source/build_dir/target-mips_24kc_musl/linux-ath79_generic/tmp/openmptcprouter-v0.59beta7-v0.59beta7-c4157bfc-r0+19602-cb1dc49c18-ath79-generic-xiaomi_aiot-ac2350-squashfs-sysupgrade.bin
84456+1 records in
84456+1 records out
43241662 bytes (43 MB, 41 MiB) copied, 0.126216 s, 343 MB/s
/home/rdmitry0911/projects/openmptcprouter/ath79/5.15/source/staging_dir/host/bin/padjffs2 /home/rdmitry0911/projects/openmptcprouter/ath79/5.15/source/build_dir/target-mips_24kc_musl/linux-ath79_generic/tmp/openmptcprouter-v0.59beta7-v0.59beta7-c4157bfc-r0+19602-cb1dc49c18-ath79-generic-xiaomi_aiot-ac2350-squashfs-sysupgrade.bin  64
padding image to 02bb0000
bash: 8: Bad file descriptor
echo '{  "metadata_version": "1.1", "compat_version": "1.0",   "supported_devices":["xiaomi,aiot-ac2350"], "version": { "dist": "openmptcprouter", "version": "v0.59beta7-v0.59beta7-c4157bfc", "revision": "r0+19602-cb1dc49c18", "target": "ath79/generic", "board": "xiaomi_aiot-ac2350" } }' | fwtool -I - /home/rdmitry0911/projects/openmptcprouter/ath79/5.15/source/build_dir/target-mips_24kc_musl/linux-ath79_generic/tmp/openmptcprouter-v0.59beta7-v0.59beta7-c4157bfc-r0+19602-cb1dc49c18-ath79-generic-xiaomi_aiot-ac2350-squashfs-sysupgrade.bin
Failed to open firmware file
make[5]: [Makefile:112: /home/rdmitry0911/projects/openmptcprouter/ath79/5.15/source/build_dir/target-mips_24kc_musl/linux-ath79_generic/tmp/openmptcprouter-v0.59beta7-v0.59beta7-c4157bfc-r0+19602-cb1dc49c18-ath79-generic-xiaomi_aiot-ac2350-squashfs-sysupgrade.bin] Error 1 (ignored)
sha256sum "/home/rdmitry0911/projects/openmptcprouter/ath79/5.15/source/build_dir/target-mips_24kc_musl/linux-ath79_generic/tmp/openmptcprouter-v0.59beta7-v0.59beta7-c4157bfc-r0+19602-cb1dc49c18-ath79-generic-xi [... cut ...] openmptcprouter/ath79/5.15/source/bin/targets/ath79/generic/openmptcprouter-v0.59beta7-v0.59beta7-c4157bfc-r0+19602-cb1dc49c18-ath79-generic-xiaomi_aiot-ac2350.manifest
make[5]: Leaving directory '/home/rdmitry0911/projects/openmptcprouter/ath79/5.15/source/target/linux/ath79/image'
make[4]: Leaving directory '/home/rdmitry0911/projects/openmptcprouter/ath79/5.15/source/target/linux/ath79'
make[3]: Leaving directory '/home/rdmitry0911/projects/openmptcprouter/ath79/5.15/source/target/linux'

```

Ysurac, 2022-07-03:
You can remove some big part like v2ray that is quite huge and that will not  work on a platform with low CPU/Memory.

rdmitry0911, 2022-07-03:
> You can remove some big part like v2ray that is quite huge and that will not work on a platform with low CPU/Memory.

How to safely remove unneeded dependencies (v2ray,  omr-bypass etc) from omr to reduce footprint? Now I remove them  manually editing .config file after "make menuconfig", but I'm not sure I do this correctly.

closed 2022-10-06 by github-actions[bot]

HSTelecom, 2022-12-01:
Hi
please can you share .conf or bin file for Mi R2350 IOT 
I tried to build it  with ubuntu PC take 3hr making attached file , I tried to upload it via OPEN-WRT steps SCP/flash script  with fail on update still on orange led.

I tried with OPEN-WR flashing method and web update both working fine.

also I want to bulid it for mikrotik RB2011L and Ubnt RouterStation all are based on AT73xx which its not found on make menu , only AT79xx.

Thanks for your support in advance.