Ysurac/openmptcprouter #2416: Performance of scheduler default, ecf and roundrobin issue

guohaorui, 2022-07-06:
## Expected Behavior

I have a topology like below to test the performance of MPTCP when middle links are added packet lost. There are 4 sublinks where the packet loss rate will be manually added using `tc` command. It is obvious that the performance of the case adding packet lost should be worse than not. 
The concrete `tc` command is `tc qdisc add dev ens9f2 root netem delay 20ms rate 20Mbit loss 0.1%`
The topology is:
[image]

## Current Behavior

There are two cases `a02` and `11`, where all configurations of `11` are same as `a02` except 4 sublinks add up packet lost using command as above. And when we look at the good throughtput, we can see that `11` performs better which is not unreasonable. And when we monitor the middle 4 sublinks, actually there are only 3 links used in `a02` case while in `11` all links are used. Besides, the system log can't give me useful reason related to it. This phenomenon confuses me and i would appreciate it if you can answer me if there is some configuration wrong.  
[image]

[image]

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.58.5 r0+16336-b36068d35d / LuCI Master git-21.077.59320-d1bf56d
  - OpenMPTCProuter VPS version: 0.1026
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform: x86

The installation command is `wget -O - https://www.openmptcprouter.com/server/debian10-x86_64.sh | sh`

Best Wishes!

guohaorui, 2022-07-06:
Emmmm, sorry i can't, because i install the 0.59 beta7 on 5.4 kernel at the begining, however, when all configuration are done, the source starts to generate packets, then it lost the information of network card i.e. it select eth0 successfully and after some minutes, it cannot select eth0 any more in interface of wan/lan setting while `ip addr` can still find the eth0, maybe it's a bug?

Ysurac, 2022-07-06:
I don't understand what you mean.
You start from a fresh config ?

guohaorui, 2022-07-06:
Here's how it comes. When i flash the 0.59 beta7 on 5.4 kernel into linux, it can select eth0 in the interface of wan/lan setting successfully. And when traffic source starts to send packets through MPTCP, the eth0 option just suddenly disappear and it can not find eth0 any more while actually eth0 can be detected by `ip addr` command

guohaorui, 2022-07-06:
[image]

guohaorui, 2022-07-07:
Hi, is there any possible way to see the concrete scheduler's decision e.g. for roundrobin, 0-1000ns select the first link and so on, where i can find out why only three links of four is actually functioning

guohaorui, 2022-07-15:
Hi, currently i am able to update the version to v0.59beta8 and it can normally transmit packets, but still there is one problem: when intermediate links add 0.01% packet loss rate, the good throughput is higher than not adding 0.1% packet loss rate.
This is important for me and it would be appreciated if you can help me solve it.

closed 2022-10-18 by github-actions[bot]