Ysurac/openmptcprouter #2417: Outbound Connection Attempts from VPS

lukeiggle, 2022-07-07:
## Expected Behavior

Expectation would be that VPS would not try to TCP SYN the MP-Router IP as they are usually always a CGNAT address, or RFC1918 address behind a Public IPv4 address therefore not reachable via the internet.

## Current Behavior

Huge amounts of constant attempts to connect to all 3 x IP addresses which are bound to the MP-Router.

## Possible Solution

Unknown, unsure how/why this is happening

## Steps to Reproduce the Problem

  1. on VPS - netstat -an | grep SYN_SENT
  2. or tcpdump on VPS host and look for outbound SYN connections to any IP's that are directly bound to a MP-router

## Context (Environment)

Back chatter leaving half open TCP SYN's.  Understanding if this is mis-configuration, or what this is trying to do?

Isolated the MP-Router this was consuming a couple of hundred kbit/s at times just sitting idle.

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.59beta6-5.4 r0+16594-ce9
  - OpenMPTCProuter VPS version:  5.4.196-mptcp 0.1027-test
  - OpenMPTCProuter VPS provider: Bare metal server in a DC.
  - Country: Australia

Pastebin of the netstat -an | grep SYN_SENT on my VPS:
https://pastebin.com/bkSj9QmM

Ysurac, 2022-07-07:
I don't have this issue.
What is the proxy/vpn used ? the defaults ones ?
Did you made any change in the configuration ?

lukeiggle, 2022-07-07:
I’m using shadow socks and glorytun tcp. 

Don’t think many things have changed in the config.

 I’m using bbr for congestion control, that’s about it. 
 
There is sqm on the adsl interface. 

Other than that pretty standard.

lukeiggle, 2022-07-10:
Is there a log somewhere for glorytun tcp that shows connection attempts and disconnects?

How can I tune glorytun to retry TCP connections if they drop out?

lukeiggle, 2022-07-11:
Are we able to provide some more information on this one to get it looked at?

lukeiggle, 2022-07-18:
`Chain OUTPUT (policy DROP 0 packets, 0 bytes)
 pkts bytes target     prot opt in     out     source               destination
  281 20028 REJECT     tcp  --  any    enp1s0f0  anywhere             VPS-IP  tcp flags:SYN,ACK/SYN state NEW reject-with icmp-port-unreachable
  434 31248 REJECT     tcp  --  any    enp1s0f0  anywhere             100.64.0.0/10        reject-with icmp-port-unreachable
  435 31320 REJECT     tcp  --  any    enp1s0f0  anywhere             10.0.0.0/24          reject-with icmp-port-unreachable`

100.64.0.0/10 = CGNAT range, used by starlink (un-routeable on the internet)
10.0.0.0/24 = Inside ADSL range

I've added the above IP tables. As you can see even after a few minutes there are a lot of packets are bytes trying to reach the INSIDE IP's of all 3 connections. I'd really like to help solve this for you but don't know where to start.

Since I put these commands in, the ADSL interface has become more reliable. I witnessed a significant amount of ramp up of open TCP connections, potentially exhausting the ADSL router, or the VPS.

@Ysurac  where can I put these iptables custom commands so they don't get flushed when we reload?

`iptables --insert OUTPUT 1 -o enp1s0f0 --protocol tcp --dst 10.0.0.10/24 --jump REJECT
iptables --insert OUTPUT 1 -o enp1s0f0 --protocol tcp --dst 100.64.0.0/10 --jump REJECT
iptables --insert OUTPUT 1 -o enp1s0f0 --protocol tcp --tcp-flags SYN,ACK SYN -m state --state NEW --dst LTE-IP --jump REJECT`

Note, the LTE one has a dedicated IPV4 address so I've had to change the state on this command so it doesn't break connectivity.

Cheers

closed 2022-10-22 by github-actions[bot]