Ysurac/openmptcprouter #2418: Glorytun TCP Dropout Starlink (CGNAT)

lukeiggle, 2022-07-07:
## Expected Behavior

I have 3 x WAN Connections. Starlink / LTE / ADSL

Glorytun TCP should have connections bound to all 3 x WAN interfaces.

On first restart of Glorytun TCP we can see all 3 x TCP Sessions established:

netstat -an | grep 65001
tcp6       0      0 :::65001                :::*                    LISTEN
tcp6       0      0 VPS:65001      STARLINK-OUTSIDE-IP:42458    ESTABLISHED
tcp6       0      0 VPS:65001      LTE-DIRECT-IP:21573     ESTABLISHED
tcp6       0      0 VPS:65001      ADSL-OUTSIDE-IP:50882    ESTABLISHED

## Current Behavior

Minutes after a restart we see the connection for the Starlink service has dropped out and never re-establishes:
netstat -an | grep 65001
tcp6       0      0 :::65001                :::*                    LISTEN
tcp6       0      0 VPS:65001      LTE-DIRECT-IP:21573     ESTABLISHED
tcp6       0      0 VPS:65001      ADSL-OUTSIDE-IP:50882    ESTABLISHED

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.59beta6-5.4 r0+16594-ce9
  - OpenMPTCProuter VPS version: 5.4.196-mptcp 0.1027-test
  - OpenMPTCProuter VPS provider: Bare Metal in a DC
  - OpenMPTCProuter platform: VPS on x64 Hypervisor

Ysurac, 2022-07-07:
What do you have in router part system log ?

lukeiggle, 2022-07-07:
Nothing comes up or logs around the time that the tcp session drops on the cgnat starlink interface. 

Occasionally the vpn drops and re-establishes, but this could be because of it just being on the adsl and lte interface.

lukeiggle, 2022-07-10:
Is there a log somewhere for glorytun tcp that shows connection attempts and disconnects?

How can I tune glorytun to retry TCP connections if they drop out?

bdruth, 2022-07-17:
Interesting, I'm seeing the same on my side w/ Starlink.

Ysurac, 2022-07-17:
Not sure if it's the same issue. You only have Starlink as connection ? if not what is the connection set as Master ?

You can try an other VPN in System->OpenMPTCProuter, "advanced settings" checkbox like Glorytun UDP.

bdruth, 2022-07-17:
I have VDSL + Starlink, so similar, just 2 connections instead of 3. Switched over to DSVPN and it took longer, but when I checked back after a couple hours, similar - the Starlink side of the VPN was down. Also, nothing notable in the DSVPN logs ... though, there's a strange error about `RTNETLINK` that doesn't appear to impact anything?

```
2022-07-17T09:10:29-05:00 OpenMPTCProuter dsvpn: starting dsvpn vpn instance vpn
2022-07-17T09:10:29-05:00 OpenMPTCProuter dsvpn[1303]: Interface: [tun0]
2022-07-17T09:10:29-05:00 OpenMPTCProuter dsvpn[1303]: Trying to reconnect
2022-07-17T09:10:29-05:00 OpenMPTCProuter dsvpn[1303]: Connecting to VPS-IP:65401...
2022-07-17T09:10:29-05:00 OpenMPTCProuter dsvpn[1303]: RTNETLINK answers: Permission denied
2022-07-17T09:10:29-05:00 OpenMPTCProuter dsvpn[1303]: Connected
2022-07-17T12:24:52-05:00 OpenMPTCProuter dsvpn[1303]: Done.
2022-07-17T12:24:54-05:00 OpenMPTCProuter dsvpn: starting dsvpn vpn instance vpn
2022-07-17T12:24:54-05:00 OpenMPTCProuter dsvpn[3213]: Interface: [tun0]
2022-07-17T12:24:54-05:00 OpenMPTCProuter dsvpn[3213]: Trying to reconnect
2022-07-17T12:24:54-05:00 OpenMPTCProuter dsvpn[3213]: Connecting to VPS-IP:65401...
2022-07-17T12:24:54-05:00 OpenMPTCProuter dsvpn[3213]: RTNETLINK answers: Permission denied
2022-07-17T12:24:54-05:00 OpenMPTCProuter dsvpn[3213]: Connected
2022-07-17T12:24:54-05:00 OpenMPTCProuter dsvpn[3213]: Done.
2022-07-17T12:24:56-05:00 OpenMPTCProuter dsvpn: starting dsvpn vpn instance vpn
2022-07-17T12:24:56-05:00 OpenMPTCProuter dsvpn[3942]: Interface: [tun0]
2022-07-17T12:24:56-05:00 OpenMPTCProuter dsvpn[3942]: Trying to reconnect
2022-07-17T12:24:56-05:00 OpenMPTCProuter dsvpn[3942]: Connecting to VPS-IP:65401...
2022-07-17T12:24:56-05:00 OpenMPTCProuter dsvpn[3942]: RTNETLINK answers: Permission denied
2022-07-17T12:24:56-05:00 OpenMPTCProuter dsvpn[3942]: Connected
```
Somewhere betw 9:10am and when I checked around 12:20pm is when it dropped the Starlink side.
```
OpenMPTCProuter# netstat -an | grep 65401
tcp        0      0 VDSL-IP:48669     VPS-IP:65401      ESTABLISHED
tcp        0      0 Starlink-IP:14394    VPS-IP:65401      ESTABLISHED
```

bdruth, 2022-07-17:
Few hours later, DSVPN is still maintaining both connections ...
```
OpenMPTCProuter# netstat -an | grep 65401
tcp        0      0 VDSL-IP:48669     VPS-IP:65401      ESTABLISHED
tcp        0      0 Starlink-IP:14394    VPS-IP:65401      ESTABLISHED
```

lukeiggle, 2022-07-17:
@bdruth Thanks for responding. I'm trying DSVPN now.

Wondering if you are seeing the same issues I'm seeing on issue  #2417 as well?

lukeiggle, 2022-07-18:
Perhaps 30 minutes later DSVPN stops on starlink for me as well.

VPS# netstat -an | grep 65401
tcp        0      0 0.0.0.0:65401           0.0.0.0:*               LISTEN
tcp        0      0 VPS-IP:65401      LTE:19964     ESTABLISHED
tcp        0      0 VPS-IP:65401      VDSL:59148    ESTABLISHED

bdruth, 2022-07-18:
I've seen it stay up longer on DSVPN, but it definitely does drop eventually. I'm wondering why it doesn't re-establish once it drops, that's the real question. Overall, my Starlink stays connected quite well (no obstructions, etc.), so it's odd that it's consistently that connection that's dropped and not re-established.

bdruth, 2022-07-18:
I don't know that I have the #2417 issue @lukeiggle - at least, not that I'm aware of - I don't see any connections when I use the diagnostic you posted - `netstat -an | grep SYN_SENT` from my VPS.

Ysurac, 2022-07-18:
As I said, did you try Glorytun UDP ?

Ysurac, 2022-07-18:
Can you also try to set Network->MPTCP, "Multipath TCP scheduler" to "Default" ?
It seems that I have same issue with BLEST but not with default scheduler.

bdruth, 2022-07-18:
OK - will get that sorted shortly.

bdruth, 2022-07-18:
OK, switched to default instead of `BLEST` - also switched to Glorytun UDP - but I'm seeing this in the logs:
```
2022-07-18T13:33:13-05:00 OpenMPTCProuter glorytun-udp: starting glorytun-udp vpn instance vpn
2022-07-18T13:33:13-05:00 OpenMPTCProuter glorytun-udp[15897]: running on device tun0 as pid 15897
2022-07-18T13:33:36-05:00 OpenMPTCProuter glorytun-udp: starting glorytun-udp vpn instance vpn
2022-07-18T13:33:36-05:00 OpenMPTCProuter glorytun-udp[20781]: running on device tun0 as pid 20781
2022-07-18T13:33:54-05:00 OpenMPTCProuter glorytun-udp: starting glorytun-udp vpn instance vpn
2022-07-18T13:33:54-05:00 OpenMPTCProuter glorytun-udp[24430]: running on device tun0 as pid 24430
2022-07-18T13:34:40-05:00 OpenMPTCProuter glorytun-udp: starting glorytun-udp vpn instance vpn
2022-07-18T13:34:40-05:00 OpenMPTCProuter glorytun-udp[1278]: running on device tun0 as pid 1278
2022-07-18T13:34:54-05:00 OpenMPTCProuter glorytun-udp: starting glorytun-udp vpn instance vpn
2022-07-18T13:34:54-05:00 OpenMPTCProuter glorytun-udp[4045]: running on device tun0 as pid 4045
2022-07-18T13:35:30-05:00 OpenMPTCProuter glorytun-udp: starting glorytun-udp vpn instance vpn
2022-07-18T13:35:30-05:00 OpenMPTCProuter glorytun-udp[10486]: running on device tun0 as pid 10486
2022-07-18T13:35:44-05:00 OpenMPTCProuter glorytun-udp: starting glorytun-udp vpn instance vpn
2022-07-18T13:35:44-05:00 OpenMPTCProuter glorytun-udp[12892]: running on device tun0 as pid 12892
```

bdruth, 2022-07-18:
It doesn't seem like it's liking Glorytun UDP - 
[image]

Ysurac, 2022-07-18:
Possible Glorytun UDP was not working on 0.57.3
The Multipath Scheduler change is only for TCP VPN like Glorytun TCP.

bdruth, 2022-07-18:
Switching back to Glorytun TCP and the connection comes up - will see if it keep the Starlink iface w/ the MP scheduler change.

bdruth, 2022-07-18:
Still up - been more than a couple minutes, the `BLEST` MP scheduler change might be the trick.

lukeiggle, 2022-07-18:
I changed to DEFAULT on the MP Scheduler (I had BLEST), switched to glorytun TCP.

Did a speedtest.net - and got one of my fastest speed tests ever. Checked glorytun-tcp and it had dropped out within a few minutes.. perhaps the speed test exacerbated it?

Switched to DSVPN, ran the speedtest a couple of times. Still up, will report back.

bdruth, 2022-07-18:
Interesting - I ran a speedtest just to replicate and mine's still up ... 🤷‍♂️

lukeiggle, 2022-07-19:
Glorytun-tcp still drops. for the record, glorytun-udp does the same on starlink.

My best guess is due to the nature of starlink and hopping between satellites the CGNAT binding times out or gets a TCP-RESET and kills the connection. Then Glorytun-tcp then doesn't try and reconnect. Is there something we can parse to glorytun-tcp to keepalive or reconnect sessions without sending a HUP and tearing down the VPN and interrupting traffic flow?

So much traffic like voice/teams etc relies on UDP.. to have the main connection out of 3 down for the tunnel makes this almost unusable.

GrantSparks, 2022-07-19:
One of my connections is starlink and it seems stable enough.  v0.58.5.   I don't have any expert insight but I'd be happy to compare settings or try things out if it can help debug issues.

Ysurac, 2022-07-19:
With default scheduler I don't have the issue. I also have a starlink connection. When the connection is down for any reason, the subflow is removed and added back.
For the drop during speedtest you have to set SQM with download and upload speed.

lukeiggle, 2022-07-19:
@Ysurac given the fluctuating nature of starlink, what would speeds you recommend setting sqm for this interface?

Is there any log for tcp drops on glorytun?

are you running the latest beta build like me as well?

Ysurac, 2022-07-19:
@lukeiggle On latest beta/snapshot, there is an autorate function for SQM, so you set max download and upload speed and check this box.
There is no detailed logs for glorytun.
I'm running latest snapshot build (and latest release).

bdruth, 2022-07-19:
I'm on 0.57.3, so not latest betas here (maybe I should upgrade?) - w/ default scheduler I do still get the Starlink connection dropped and not re-established on Glorytun-TCP ... it just seems to take longer than a couple minutes.

bdruth, 2022-07-19:
kk, I'll give it a shot - I had tried 58.5 a while back but had [issues with UDP traffic](https://github.com/Ysurac/openmptcprouter/discussions/2059) not being handled well (interrupting Zoom calls) - I'll give it another go as it's been awhile :)

bdruth, 2022-07-21:
switched over to latest `rc` last night ... with default MP scheduler - was downloading the macOS 12.5 update and my wife noted that nothing was loading for her ... flipped the MP scheduler over to BLEST (still downloading) and she said that everything started working again. I do have SQM set on my DSL (which has stable up/down speeds) and set SQM autorate on the Starlink ... unfortunately, I can't use the default MP scheduler if it's going to cause issues with other connections when I'm saturating the links. I haven't checked Glorytun yet today, but based on earlier indications, but I think if I'm running BLEST I'm going to get the same behavior?

bdruth, 2022-07-21:
Might be unrelated, but just had a complete failure of the VPN on latest RC - ended up having to restart Glorytun on the VPS ... 
```
2022-07-21T13:10:00-05:00 OpenMPTCProuter glorytun[24596]: VPS-IP.65001: shutdown
2022-07-21T13:10:00-05:00 OpenMPTCProuter glorytun: starting glorytun vpn instance vpn
2022-07-21T13:10:00-05:00 OpenMPTCProuter glorytun[24596]: STOPPED tun0
2022-07-21T13:10:00-05:00 OpenMPTCProuter glorytun[20077]: INITIALIZED tun0
2022-07-21T13:10:00-05:00 OpenMPTCProuter glorytun[20077]: VPS-IP.65001: connected
2022-07-21T13:10:00-05:00 OpenMPTCProuter glorytun[20077]: STARTED tun0
2022-07-21T13:10:03-05:00 OpenMPTCProuter glorytun[20077]: VPS-IP.65001: shutdown
2022-07-21T13:10:03-05:00 OpenMPTCProuter glorytun: starting glorytun vpn instance vpn
2022-07-21T13:10:03-05:00 OpenMPTCProuter glorytun[20077]: STOPPED tun0
2022-07-21T13:10:03-05:00 OpenMPTCProuter glorytun[23154]: INITIALIZED tun0
2022-07-21T13:10:04-05:00 OpenMPTCProuter glorytun[23154]: VPS-IP.65001: connected
2022-07-21T13:10:04-05:00 OpenMPTCProuter glorytun[23154]: STARTED tun0
2022-07-21T13:10:16-05:00 OpenMPTCProuter glorytun[23154]: STOPPED tun0
2022-07-21T13:10:17-05:00 OpenMPTCProuter glorytun[23154]: VPS-IP.65001: connected
2022-07-21T13:10:17-05:00 OpenMPTCProuter glorytun[23154]: STARTED tun0
2022-07-21T13:10:17-05:00 OpenMPTCProuter glorytun[23154]: STOPPED tun0
2022-07-21T13:11:47-05:00 OpenMPTCProuter glorytun[23154]: VPS-IP.65001: connected
2022-07-21T13:11:47-05:00 OpenMPTCProuter glorytun[23154]: STARTED tun0
2022-07-21T13:11:47-05:00 OpenMPTCProuter glorytun[23154]: STOPPED tun0
2022-07-21T13:13:17-05:00 OpenMPTCProuter glorytun[23154]: VPS-IP.65001: connected
2022-07-21T13:13:18-05:00 OpenMPTCProuter glorytun[23154]: STARTED tun0
2022-07-21T13:13:18-05:00 OpenMPTCProuter glorytun[23154]: STOPPED tun0
2022-07-21T13:14:03-05:00 OpenMPTCProuter glorytun: starting glorytun vpn instance vpn
2022-07-21T13:14:03-05:00 OpenMPTCProuter glorytun[25644]: INITIALIZED tun0
2022-07-21T13:14:48-05:00 OpenMPTCProuter glorytun[25644]: VPS-IP.65001: connected
2022-07-21T13:14:48-05:00 OpenMPTCProuter glorytun[25644]: STARTED tun0
2022-07-21T13:14:49-05:00 OpenMPTCProuter glorytun[25644]: read: Connection reset by peer
2022-07-21T13:14:49-05:00 OpenMPTCProuter glorytun[25644]: STOPPED tun0
2022-07-21T13:15:00-05:00 OpenMPTCProuter glorytun[25644]: getpeername: Socket not connected
2022-07-21T13:15:00-05:00 OpenMPTCProuter glorytun: starting glorytun vpn instance vpn
2022-07-21T13:15:00-05:00 OpenMPTCProuter glorytun[3434]: INITIALIZED tun0
2022-07-21T13:15:37-05:00 OpenMPTCProuter glorytun[3434]: VPS-IP.65001: connected
2022-07-21T13:15:37-05:00 OpenMPTCProuter glorytun[3434]: STARTED tun0
2022-07-21T13:15:37-05:00 OpenMPTCProuter glorytun[3434]: STOPPED tun0
2022-07-21T13:17:02-05:00 OpenMPTCProuter glorytun: starting glorytun vpn instance vpn
2022-07-21T13:17:02-05:00 OpenMPTCProuter glorytun[21752]: INITIALIZED tun0
2022-07-21T13:17:07-05:00 OpenMPTCProuter glorytun[21752]: VPS-IP.65001: connected
2022-07-21T13:17:07-05:00 OpenMPTCProuter glorytun[21752]: STARTED tun0
```

Restarting on the OMR side didn't help via `/etc/init.d/glorytun restart`

On the VPS, I was seeing this
```
Jul 21 13:10:16 debian-09 OMR-Service: No answer from VPN client end, restart Glorytun-TCP
```

I ended doing a `systemctl restart glorytun-tcp@tun0`, which hung for awhile, until I also restarted Glorytun on the OMR side (again) - and then it seemed like everything sync'd up again.

Not sure if there's any other logs that might shed more light on this.

bdruth, 2022-07-21:
Interestingly, it's been a few hours and it looks like Glorytun still has both connections up ... even though I'm using `BLEST`

lukeiggle, 2022-07-24:
@bdruth  how's it holding up?

I've set SQM on, with no specific speeds (0) on both the LTE and the Starlink service. I've set cake and layer cake as the qos settings, but I can't find the autorate-ingress setting. Where is that?

bdruth, 2022-07-24:
@lukeiggle  seems to be going ok. Not sure about your autorate question, it's just a checkbox in my SQM QoS settings page.

lukeiggle, 2022-07-26:
Don't think I'm going crazy here:
https://ibb.co/3cZnNjz

@Ysurac  no SQM autorate on Firmware Version：openmptcprouter v0.59beta6-5.4 r0+16594-ce92de8c8c

lukeiggle, 2022-09-05:
Confirming that the latest version of 0.59.1 still has issues for me.

I've turned on SQM autorate (it's there now) and still getting dropouts on starlink and the TCP connection never comes back without a full kill of glorytun. Seems to stay up anywhere from 5 mins to and hour max.

closed 2022-12-12 by github-actions[bot]