Ysurac/openmptcprouter #2421: v2ray & shadowsocks PORTFORWARDING RULES printed to specifc IP

simonemessina92, 2022-07-07:
v0.59beta8-5.4 r0+16594
VPS version v0.59beta8 - ARUBA
x64 pc platform

I've got this problem witch is really insane. I basically OMR to do SRT / RTMP streams, I'm using v2ray proxy and no VPN at all. I followed the guide to correctly port forward with v2ray, I've got 2 pc I want to been able to stream outside the LAN.
I've basically open from port 10001-10008 on the ip address of the first PC  that I gave Ip reservation at 192.168.100.100.
Strange thing is that I can recieve all the streams on the reciever device but not the one coming from port 10007, no matter what I try it doesn't work. 
I was doing some testing and apply that pfwd rule on port 10007 to the second pc that have ip reservation at 192.168.100.150, and from there it works. It seems like even if change the port forwarding rule in the firewall tab it doesn't really change that rule, is something like that rule is "printed" to the second pc Ip address, no matter if I delete that rule, change the Ip and so on....

I don't know what to do, I've check on the VPS with cat /etc/shorewall/rules and I've got the following result: 
[image]
Here is the settings for the firewall rules 
[image]
and here's the detail I've followed for every rules ( from 10001 to 10008 )
[image]
logging to the router in SSH i run uci show firewall and see the following (for the 10007 and all the other rules)

firewall.@redirect[6]=redirect
firewall.@redirect[6].target='DNAT'
firewall.@redirect[6].name='7'
firewall.@redirect[6].proto='udp'
firewall.@redirect[6].src='vpn'
firewall.@redirect[6].src_dport='10007'
firewall.@redirect[6].dest='lan'
firewall.@redirect[6].dest_port='10007'
firewall.@redirect[6].v2ray='1'
firewall.@redirect[6].dest_ip='192.168.100.100'

Everything seems to be done correctly but the only way to recieve the signal is to stream from that second PC with IP 192.168.100.150 on port 10007, weird stuff also is that the rules works both when put "ANY" or the "IP ADDRESS"(192.168.100.150) in the tab bar of the rule. 
As I said is something like that port is redirecting traffic only from that IP address, I really don't know what else to check . I've also tried to press Restart Firewall in the status bar but nothing changed.

Can please some one explain me what else can I try to solve this problem? I wish I can easily add port forwarding to my client devices, It might happen that I've to add  rules for SRT traffic day by day depending what hardware / pc device I'm using... So I thinkg this things shouldn't happen so far.

Here's a screenshot while I transmit from the PC 2 (192.168.100.150) to pc 1 using SRT on port 10007
[image]

Can some one help me troubleshooting this ? I don't know if it's a bug or else, I've first tried with beta 7 and this problem occurred so I've tried to update to beta 8, nothing had changed. I furthermore tried to install vps again from scratch, and even I flash a clean install of the OMR image start from the beginning again, but still have this problem, even after a clean install on both VPS and Router...

simonemessina92, 2022-07-08:
I've also tested with clean install both vps and server on stable v.0.58.5 still same problem... ports stays locked to the first ip address you settle in the rule... Please really need to solve this? is there some way to restart v2ray o try to inspect it ? If I create a rule that point a specific Ip in the lan than I couldn't change it so far, the SRT signal won't work if I change that Ip to antoher device in the Lan, this is abnormal. I tried with Glorytun UDP and there's no problem like this, but unfortunately using that won't allow me to use vmix call. Please some one help me on this really need to solve :(

Ysurac, 2022-07-08:
Please use the template when you open an issue.
For v2ray, opened ports are in /etc/v2ray/v2ray-server.json

Ysurac, 2022-07-08:
I found the error, it's due to a typo. Rules in v2ray-server.json were never removed.

simonemessina92, 2022-07-08:
> I found the error, it's due to a typo. Rules in v2ray-server.json were never removed.

Sorry if I didn't post correctly using the template 🥲 so I'm a right there's something that you have to modified in order to make it work ? Sorry I'm not an IT expert , I've test it for more than 7 hours and yes it really seems as the v2ray not removed the rules. Thank you for your kind answer🙏

simonemessina92, 2022-07-09:
> I found the error, it's due to a typo. Rules in v2ray-server.json were never removed.

[image]

It's actually as you said!  If I see the web gui the rule is made up on the ip 192.168.100.218; but on the vps if I do cat /etc/v2ray/v2ray-server.json I see that the ip 192.168.100.100 is printed over that rule. Just wondering if you will modify this in order to solve this problem . May I wait for another beta or there's some command that can be added on that.json to solve this ? Sorry if I bother you, I just need to have this feature of resetting rules working by wednesday somehow because I've got a big event and I don't know how many pc they will want to stream SRT and on which ip so far. I will discover everything on the venue :(  thank you in advance for your support

Ysurac, 2022-07-11:
The issue is solved in VPS script, but you have to remove unused previous block manually. It's in 2 places in the file, you can search them using the tag (it's openmptcprouter_redir_<protocol>_<port>)

simonemessina92, 2022-07-11:
> If you clean install VPS with beta VPS script this will work.

thank you so much! I will try right now and I'll let you know if everything work!

simonemessina92, 2022-07-11:
> The issue is solved in VPS script, but you have to remove unused previous block manually. It's in 2 places in the file, you can search them using the tag (it's openmptcprouter_redir__)

I've did a clean install on both VPS and usb stick with v.059 beta8; nothing had changed so far. I've got the same error. When I create the rule with a specific Ip it stays "printed" in the json file. I've tried to modify the rule but hadn't work, I've even tried to remove the rule but it was always present in the vps. Even when unthick the rule it seems it works always on the first IP address I've specified. Have I done something wrong? Or there's still something missing? I attach the screenshot of the tests.
[image]
[image]

What I can say is that on glorytun UDP or TCP this won't happen so far, but as I said if I swap between shadowsocks and v2ray I'm encourring some problem using some apps, so I wish I can stay with v2ray that I know works the best for my needs.

Even if I swap between those, by the way, I can't no more use that port I've used in v2ray, because as I said seems printed in the vps somehow, as you said it seems no reset happen on the vps side .

Can you please let me know what else can be done in order to solve this ? Or if I might had done something wrong, the vps was a clean install on debian 10, and the usb of course was a clean install too using balena etcher on the latest beta version avaliable right now.

Ysurac, 2022-07-11:
Did you use VPS beta script ? removing/disabling a rule should work now.
You can't modify a V2Ray rule for now, this need to be removed and added again.

simonemessina92, 2022-07-11:
> Did you use VPS beta script ? removing/disabling a rule should work now. You can't modify a V2Ray rule for now, this need to be removed and added again.

I'm actually had done a clean install as I said using the beta script , the one in the snapshot here:
[image]

I thought you ment you've updated the script to been able to do this directly from the webGUI of OMR.
I was always able to delete or uncheck the rules in the firewall tab, but as you've seen in previous answer it seems that not only the rules been printed, but also while unchecked or deleted the rules are always present and locked to that specific IP.
Unfortunately this is really a problem for me, just wonder do you think this can be solved some how in the next beta or it's something that it isn't even in plan to be updated ? Just to better understand the situation, I mean it's a problem on v2ray itself or it will be solved sooner or later?

simonemessina92, 2022-07-13:
@Ysurac  So any further news about it ? What I would kindly known is if this problem is due to v2ray limitation or you think that you can add the fix in next beta version for example... I mean is it possible to expect v2ray work normally as shadowsocks+glorytun in the firewall compartment or in genaral isn't achievable ?

simonemessina92, 2022-07-19:
UPDATE to OpenMPTCProuter v0.59 rc1 #2077 

Now it seems the problem is been solved by @Ysurac ! I've tested many times switching pc and rules and actually see the VPS that refresh the rules and seems to work pretty great. Thank you