Ysurac/openmptcprouter #2423: Add Transport type `KCP` (reed-soloman encoding) configuration support for v2ray

aderk, 2022-07-09:
**Is your feature request related to a problem? Please describe.**
My problem is spiky lag (3s) for UDP. v2ray as a UDP omr proxy reduces this problem.

**Describe the solution you'd like**
v2ray has a [modified KCP](https://www.v2ray.com/en/configuration/transport/mkcp.html) option for `Transport`.

**Describe alternatives you've considered**
Trying all kinds of proxy / vpn options in OMR

**Additional context**
Thank you!

Ysurac, 2022-07-09:
mKCP is UDP based, this can't be used in OMR for aggregation.

suyuan168, 2022-07-09:
> **Is your feature request related to a problem? Please describe.** My problem is spiky lag (3s) for UDP. v2ray as a UDP omr proxy reduces this problem.
> 
> **Describe the solution you'd like** v2ray has a [modified KCP](https://www.v2ray.com/en/configuration/transport/mkcp.html) option for `Transport`.
> 
> **Describe alternatives you've considered** Trying all kinds of proxy / vpn options in OMR
> 
> **Additional context**
> 
> > KCP is a fast and reliable protocol that can achieve the transmission effect of a reduction of the average latency by 30% to 40% and reduction of the maximum delay by a factor of three, at the cost of 10% to 20% more bandwidth wasted than TCP. It is implemented by using the pure algorithm, and is not responsible for the sending and receiving of the underlying protocol (such as UDP), requiring the users to define their own transmission mode for the underlying data packet, and provide it to KCP in the way of callback. Even the clock needs to be passed in from the outside, without any internal system calls.
> 
> Thank you!

This is like not being able to integrate tinyfecVPN.
They are sent using UDP and there is no way to implement multiplexing in OMR.

aderk, 2022-07-26:
Gotcha, thanks!

closed 2022-07-26 by aderk