Ysurac/openmptcprouter #2425: Configure a VPN to access my LAN network

ElMajor76, 2022-07-11:
## Expected Behavior
I'm looking up to configure a VPN to access my LAN network. I read in some forums that's a good way to configure a VPN directly on the VPS. 
Do you have some help to give it to me for configure the VPN on my VPS ?

## Current Behavior
Actually I use a OpenVPN on a server at home to access my LAN network but the latency is horrible (around 200-300ms). If I use RDP on a VM, it's not possible without lags.
## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.58.5 r0+16336-b36068d35d
  - OpenMPTCProuter VPS version: OpenMPTCProuter VPS 0.1026
  - OpenMPTCProuter VPS provider: FirstHeberg
  - OpenMPTCProuter platform: RPI4 4GB

arahina, 2022-07-12:
Just a thought ... Why not use Wireguard VPN on your VPS and then make sure there is a static route via the omrvpn on 10.255.252.1/tun0 to your internal network?  Wireguard performs wayyyyy better than OpenVPN, is simpler to setup and there are Andoid/IOS clients for Wireguard for when you're mobile.

ElMajor76, 2022-07-13:
> Just a thought ... Why not use Wireguard VPN on your VPS and then make sure there is a static route via the omrvpn on 10.255.252.1/tun0 to your internal network? Wireguard performs wayyyyy better [...]

Hello and thank you for your response !  

I would like to use the method you describe, but I'm not comfortable with the configuration of Wireguard (I've always used OpenVPN).  

Then if you can explain to me how it goes for the configuration and the implementation of the static route, I would be interested 😀

ElMajor76, 2022-07-13:
Hello and thank you for your response !  

I would like to use the method you describe, but I'm not comfortable with the configuration of Wireguard (I've always used OpenVPN).  

Then if you can explain to me how it goes for the configuration and the implementation of the static route, I would be interested 😀

arahina, 2022-07-13:
For a starting point this has useful steps on setting up Wireguard, its pretty simple...

https://youtu.be/xlyTCuWqDOg

And also this...

https://www.wireguard.com/quickstart/

For a static route it depends on the distribution of Linux on your VPS.

closed 2022-10-17 by github-actions[bot]