Ysurac/openmptcprouter #2426: Disney+ bypass

bambuca, 2022-07-14:
## Expected Behavior
If I select "disneyplus" in section "Protocols and services" under "OMR-Bypass", I expect Disney+ to work.

## Current Behavior
Disney+ is not working and displays a black page.

## Workaround
Add these domains to "Domains" section under "OMR-Bypass":
1. www.disneyplus.com
1. static-assets.bamgrid.com
1. prod-static.disney-plus.net
1. bam-sdk-configs.bamgrid.com
1. js-agent.newrelic.com
1. bam.nr-data.net
1. disneyplus.com
1. cdn.registerdisney.go.com
1. cws.conviva.com
1. d9.flashtalking.com
1. disney-portal.my.onetrust.com
1. disneyplus.bn5x.net
1. js-agent.newrelic.com
1. dssott.com
1. adobedtm.com
1. bamgrid.com
1. disney-plus.net

inspired here https://support.opendns.com/hc/en-us/articles/360037591112-Domains-to-Allow-for-Disney-Plus

## Possible Solution
Add above domains in https://github.com/Ysurac/openmptcprouter-feeds/blob/ba300f719642c045591ae9ba9ad46c90a8d57049/luci-app-omr-bypass/root/etc/uci-defaults/41_omr-bypass#L105

## Specifications

  - OpenMPTCProuter version: 0.58.5
  - OpenMPTCProuter VPS version: 0.1026 5.4 100-mptcp
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform: RPI4
  - Country: Czech Republic

Ysurac, 2022-07-17:
I think there is too many domains here. I don't think adobetm.com is needed and some others too.
I don't have a disney+ account to test what is really needed, can you try to reduce the domain list ?

bambuca, 2022-07-20:
Now it doesn't work even with this giant set of domains. I'll try to solve it somehow and then I'll post the result here...

Kalimeiro, 2022-07-21:
try to bypass ASN 11251 and ASN 398849 and ASN 22604 to a WAN.

bambuca, 2022-09-17:
So I chatted with their tech support and they added my IP to the whitelist, so I'm already up and running.

closed 2022-09-17 by bambuca