Ysurac/openmptcprouter #2428: VPS with Debian 11

arahina, 2022-07-19:
Has anyone got the VPS install script working with Debian 11? 

My VPS provider doesn't have images for Debian 10 any more, only Debian 11 and Ubuntu 20.04.

I been using Ubuntu 20.04 on the VPS but the VPS install script is doing something nasty to Grub2 or the boot sector and so once installed I can't reboot the VPS or it won't boot (not even to a Grub2 boot menu).  So the VPS works so long as I don't reboot.

I've had to rebuild the VPS a dozen times and on the verge of giving up so thought I'd try and get the VPS to work with Debian 11.

Ysurac, 2022-07-19:
Please use the template.
What is the VPS script used ? And what is the VPS provider ?
Debian scripts are more tested than Ubuntu version.

arahina, 2022-07-19:
Sorry, I'll use the template next time.

Ubuntu 20.04 seems to be rebooting now without killing the boot proccess.  

I had been using **https://www.openmptcprouter.com/server/ubuntu20.04-x86_64.sh** from the website documentation.  At today's date I think this script installs server Version 0.1026 5.4.100-mptcp 

But then from going through all the issues on Github I figured there was a newer script located at the URL **https://www.openmptcprouter.com/server-test/ubuntu20.04-x86_64.sh** .  At today's date this script installs server Version 0.1027-test 5.4.196-mptcp

I'll close this issue