Ysurac/openmptcprouter #2429: WiFi AP switches to Client from Master and does not Broadcast

chop249, 2022-07-19:
## Expected Behavior

Raspberry Pi 4B should act as a WiFi access point.

## Current Behavior

I get it all set up according to Ysurac's instructions but the SSID is not broadcast and the mode switches from Master to Client. I've tried starting from a fresh copy of OpenMPTCPRouter and it does the same. This is what I am following:

https://github.com/Ysurac/openmptcprouter/wiki/Configure-Wifi

## Specifications

  - OpenMPTCProuter version: 
  - [openmptcprouter v0.58.5 r0+16336-b36068d35d](https://github.com/ysurac/openmptcprouter)
  - OpenMPTCProuter VPS version: 
  - 0.3+20210508
  - OpenMPTCProuter VPS provider:
  - Self Hosted on a Debian 10 box
  - OpenMPTCProuter platform: 
  - RPi4B

Ysurac, 2022-07-20:
I would need the result of ```uci show wireless``` via SSH on the router or "wireless" part in System->OpenMPTCProuter->Show all config.

chop249, 2022-07-20:
wireless.radio0=wifi-device
wireless.radio0.type='mac80211'
wireless.radio0.path='platform/soc/fe300000.mmcnr/mmc_host/mmc1/mmc1:0001/mmc1:0001:1'
wireless.radio0.htmode='HT20'
wireless.radio0.hwmode='11g'
wireless.radio0.channel='auto'
wireless.radio0.txpower='20'
wireless.radio0.country='US'
wireless.radio0.cell_density='0'
wireless.radio0.disabled='1'
wireless.default_radio0=wifi-iface
wireless.default_radio0.device='radio0'
wireless.default_radio0.mode='ap'
wireless.default_radio0.skip_inactivity_poll='1'
wireless.default_radio0.ssid='******'
wireless.default_radio0.encryption='psk2'
wireless.default_radio0.key='**********'
wireless.default_radio0.network='wifi lan'
wireless.default_radio0.disabled='1'

chop249, 2022-07-20:
[image]

Ysurac, 2022-07-20:
Try to set a fixed channel and cell density and then disable and enable the interface.

chop249, 2022-07-20:
Adding the channel make it go Wireless is not associated. 

wireless.radio0=wifi-device
wireless.radio0.type='mac80211'
wireless.radio0.path='platform/soc/fe300000.mmcnr/mmc_host/mmc1/mmc1:0001/mmc1:0001:1'
wireless.radio0.htmode='HT20'
wireless.radio0.hwmode='11g'
wireless.radio0.country='US'
wireless.radio0.cell_density='1'
wireless.radio0.channel='1'
wireless.default_radio0=wifi-iface
wireless.default_radio0.device='radio0'
wireless.default_radio0.mode='ap'
wireless.default_radio0.skip_inactivity_poll='1'
wireless.default_radio0.ssid='*******'
wireless.default_radio0.encryption='psk2'
wireless.default_radio0.key='***********'
wireless.default_radio0.network='wifi lan'

chop249, 2022-07-20:
I got it up and running. I had to change it from N to Legacy. Not sure why but it worked.

closed 2022-10-24 by github-actions[bot]