Ysurac/openmptcprouter #2432: Adding latency to WAN interface

satmarin, 2022-07-22:
Hi Yannick,

I would need to add latency to some network interfaces for them to be in line with others (satellite - 4G)  
Tried setting interface - advanced - latency to 500 but that did not seem to affect anything.
Is there a specific syntax required or is this not working.

I also looked at loading opkg-netem but that didn't seem possible (yet)

This is on 0.58.5

Thanks
Michaël

Ysurac, 2022-07-22:
The latency setting use netem, so netem is installed by default.
What do you have in ```uci show network``` and what the result of ```tc qdisc show``` vis SSH on the router ?

satmarin, 2022-07-22:
I had put 500 on WAN7

root@Hyper-Gate:~# uci show network
network.loopback=interface
network.loopback.ifname='lo'
network.loopback.proto='static'
network.loopback.ipaddr='127.0.0.1'
network.loopback.netmask='255.0.0.0'
network.loopback.multipath='off'
network.loopback.macaddr='00:00:00:00:00:00'
network.loopback.metric='6'
network.globals=globals
network.globals.multipath='enable'
network.globals.mptcp_path_manager='fullmesh'
network.globals.mptcp_checksum='0'
network.globals.mptcp_debug='0'
network.globals.mptcp_syn_retries='2'
network.globals.mptcp_fullmesh_num_subflows='1'
network.globals.mptcp_fullmesh_create_on_err='1'
network.globals.mptcp_ndiffports_num_subflows='1'
network.globals.ula_prefix='fd24:0399:ff9b::/48'
network.globals.mptcp_scheduler='default'
network.globals.congestion='bbr'
network.lan=interface
network.lan.ifname='eth0'
network.lan.proto='static'
network.lan.ip6assign='60'
network.lan.delegate='0'
network.lan.multipath='off'
network.lan.ip4table='lan'
network.lan.macaddr='3c:ec:ef:02:fd:88'
network.lan.metric='7'
network.lan.defaultroute='0'
network.lan.peerdns='0'
network.lan.ipv6='1'
network.lan.ipaddr='192.168.100.1'
network.lan.netmask='255.255.255.0'
network.lan.label='Hyper-G'
network.lan.addlatency='0'
network.wan1=interface
network.wan1.proto='static'
network.wan1.ip4table='wan'
network.wan1.defaultroute='0'
network.wan1.metric='8'
network.wan1.peerdns='0'
network.wan1.ipv6='0'
network.wan1.type='macvlan'
network.wan1.masterintf='eth2'
network.wan1.label='W1'
network.wan1.gateway='10.10.0.1'
network.wan1.ifname='wan1'
network.wan1.addlatency='0'
network.wan1.macaddr='3c:ec:ef:02:fd:01'
network.wan1.ipaddr='10.10.0.2'
network.wan1.netmask='255.255.255.0'
network.wan1.multipath='master'
network.wan2=interface
network.wan2.proto='static'
network.wan2.ip4table='wan'
network.wan2.defaultroute='0'
network.wan2.metric='9'
network.wan2.peerdns='0'
network.wan2.ipv6='0'
network.wan2.type='macvlan'
network.wan2.masterintf='eth2'
network.wan2.label='W2'
network.wan2.ipaddr='10.10.1.2'
network.wan2.netmask='255.255.255.0'
network.wan2.gateway='10.10.1.1'
network.wan2.ifname='wan2'
[... 174 lines cut ...]
qdisc fq_codel 0: dev eth0 parent :2 limit 10240p flows 1024 quantum 1514 target                                                                                      5ms interval 100ms memory_limit 32Mb ecn drop_batch 64
qdisc fq_codel 0: dev eth0 parent :1 limit 10240p flows 1024 quantum 1514 target                                                                                      5ms interval 100ms memory_limit 32Mb ecn drop_batch 64
qdisc mq 0: dev eth1 root
qdisc fq_codel 0: dev eth1 parent :4 limit 10240p flows 1024 quantum 1514 target                                                                                      5ms interval 100ms memory_limit 32Mb ecn drop_batch 64
qdisc fq_codel 0: dev eth1 parent :3 limit 10240p flows 1024 quantum 1514 target                                                                                      5ms interval 100ms memory_limit 32Mb ecn drop_batch 64
qdisc fq_codel 0: dev eth1 parent :2 limit 10240p flows 1024 quantum 1514 target                                                                                      5ms interval 100ms memory_limit 32Mb ecn drop_batch 64
qdisc fq_codel 0: dev eth1 parent :1 limit 10240p flows 1024 quantum 1514 target                                                                                      5ms interval 100ms memory_limit 32Mb ecn drop_batch 64
qdisc mq 0: dev eth2 root
qdisc fq_codel 0: dev eth2 parent :4 limit 10240p flows 1024 quantum 1514 target                                                                                      5ms interval 100ms memory_limit 32Mb ecn drop_batch 64
qdisc fq_codel 0: dev eth2 parent :3 limit 10240p flows 1024 quantum 1514 target                                                                                      5ms interval 100ms memory_limit 32Mb ecn drop_batch 64
qdisc fq_codel 0: dev eth2 parent :2 limit 10240p flows 1024 quantum 1514 target                                                                                      5ms interval 100ms memory_limit 32Mb ecn drop_batch 64
qdisc fq_codel 0: dev eth2 parent :1 limit 10240p flows 1024 quantum 1514 target                                                                                      5ms interval 100ms memory_limit 32Mb ecn drop_batch 64
qdisc mq 0: dev eth3 root
qdisc fq_codel 0: dev eth3 parent :4 limit 10240p flows 1024 quantum 1514 target                                                                                      5ms interval 100ms memory_limit 32Mb ecn drop_batch 64
qdisc fq_codel 0: dev eth3 parent :3 limit 10240p flows 1024 quantum 1514 target                                                                                      5ms interval 100ms memory_limit 32Mb ecn drop_batch 64
qdisc fq_codel 0: dev eth3 parent :2 limit 10240p flows 1024 quantum 1514 target                                                                                      5ms interval 100ms memory_limit 32Mb ecn drop_batch 64
qdisc fq_codel 0: dev eth3 parent :1 limit 10240p flows 1024 quantum 1514 target                                                                                      5ms interval 100ms memory_limit 32Mb ecn drop_batch 64
qdisc noqueue 0: dev BALANCER root refcnt 2
qdisc noqueue 0: dev wan1 root refcnt 2
qdisc noqueue 0: dev wan2 root refcnt 2
qdisc noqueue 0: dev wan3 root refcnt 2
qdisc noqueue 0: dev wan4 root refcnt 2
qdisc noqueue 0: dev wan5 root refcnt 2
qdisc noqueue 0: dev wan6 root refcnt 2
qdisc fq_codel 0: dev tun0 root refcnt 2 limit 10240p flows 1024 quantum 1500 ta                                                                                     rget 5ms interval 100ms memory_limit 32Mb ecn drop_batch 64

Ysurac, 2022-07-22:
This was not working due to a typo, this will be fixed for next release.

satmarin, 2022-07-22:
Thanks Yannick,
Is there a way I can correct it manually?  or can I enter the netem instructions somewhere manually (so they are processed at reboot)
Best regards
Michaël

Ysurac, 2022-07-22:
This is the change I made: https://github.com/Ysurac/openmptcprouter-feeds/commit/debdc682e15227e4c7ce8ff40c6e35df7feb3d9c
You then need to launch ```/etc/init.d/openmptcprouter restart```

satmarin, 2022-07-22:
Thanks, I'll give it a shot.

closed 2022-10-25 by github-actions[bot]