Ysurac/openmptcprouter #2434: Unable to ping VPS server but mptcp works (but unstable and slower)

samuelchrist, 2022-07-22:
## Expected Behavior
The server should be able to ping and get a response

## Current Behavior
Getting error when trying to ping the server
[image]

## Specifications

  - OpenMPTCProuter version: v0.58.5
  - OpenMPTCProuter VPS version:  Ubuntu 20.04
  - OpenMPTCProuter VPS provider: Vultr
  - OpenMPTCProuter platform: RPI4B 8GB

samuelchrist, 2022-07-22:
@Ysurac Thank you very much for developing OMR. I am new to OMR and would need your thought on this.

Ysurac, 2022-07-22:
You can disable HTTP API test in System->OpenMPTCProuter, "advanced settings" tab "Disable server http test" checkbox
You can also increase "VPS checks timeout"

samuelchrist, 2022-07-22:
Thank you for the revert @Ysurac. I tried but got the below error. 
[image]

Ysurac, 2022-07-22:
Check Status->System log to know why it can't contact VPS API/why connections drop and check the API key used in wizard.

samuelchrist, 2022-07-22:
[image]

Ysurac, 2022-07-22:
You rebooted the VPS after install ?

samuelchrist, 2022-07-22:
Yes @Ysurac Mutiple times

Ysurac, 2022-07-22:
On the VPS, try a ```systemctl restart omr-admin```

samuelchrist, 2022-07-22:
[image]

Ysurac, 2022-07-22:
What you have in ```journalctl -xe``` and in ```/var/log/daemon.log``` ?

samuelchrist, 2022-07-22:
[image]
[image]

Ysurac, 2022-07-22:
What is the OS used on the VPS ?
I think there is a problem in ```/etc/openmptcprouter-vps-admin/omr-admin-config.json```

samuelchrist, 2022-07-22:
@Ysurac  .Its Ubuntu 20.04

Ysurac, 2022-07-22:
So it's the problem.
As it's a fresh install, you should reinstall from scratch.

samuelchrist, 2022-07-22:
Thank you @Ysurac

closed 2022-07-22 by samuelchrist