Ysurac/openmptcprouter #2435: Package Versions OpenMPTCPRouter != OpenWRT

SonicFM, 2022-07-24:
Hello all!
can someone explain to me how
https://download.openmptcprouter.com/release/v0.58.5/bpi-r2/packages/arm_cortex-a7_neon-vfpv4/base/
and
https://packages.openmptcprouter.com/v0.58.5/arm_cortex-a7_neon-vfpv4/base/
(and the other repo folders like luci,telephony, etc.) differ, are those the same?

I recently discovered that OpenMPTCPRouter's repo and OpenWRT's repo differ in terms of package updates. Is there any particular reason for this?

Since OpenMPTCPRouter is pinned to a specific commit within the master branch of OpenWRT, as stated in https://github.com/Ysurac/openmptcprouter/issues/1904, it is more or less considered a "snapshot", is that correct?

So my question is if there is a way to keep the packages more up to date, like those in the OpenWRT repo?
The kernel packages (kmods) are updated with each new version of OpenMPTCPRouter, which is good and makes sense, but the OpenMPCTPRouter unrelated packages (e.g. the "usteer" package is completely missing), are outdated sometimes.

Here is an example of a difference in the package versions:
https://packages.openmptcprouter.com/v0.58.5/arm_cortex-a7_neon-vfpv4/packages/banip_0.7.5-3_all.ipk
https://downloads.openwrt.org/releases/packages-21.02/arm_cortex-a7_neon-vfpv4/packages/banip_0.7.10-3_all.ipk

However its seems that there is kind of a sync but only when a new OpenMPTCPRouter Version is worked on, correct?
https://packages.openmptcprouter.com/v0.59rc1/arm_cortex-a7_neon-vfpv4/packages/banip_0.7.10-3_all.ipk

Currently I have "worked around" the problem by including the OpenWRT feeds from release 21.02.

I would be very pleased to receive an answer to my questions.

Sorry for the possibly "stupid" questions, I'm just trying to understand better.

Thanks for your great work on OpenMPTCPRouter, Ysurac!

Ysurac, 2022-07-24:
packages in packages.openmptcprouter.com are a snapshot of OpenWRT packages.
packages in download.openmptcprouter.com are build and used in the images.

This way I can update snapshot of OpenWRT packages if I want and keep original packages.

OpenWRT packages are not automatically updated because this can make some conflict with OpenMPTCProuter packages.

But you are free to do what you want.

SonicFM, 2022-07-24:
Thanks for the quick reply!
Just wanted to understand the differences and reasons, thank you! :)

closed 2022-07-24 by SonicFM